tiny-pilot/tinypilot #789: X button sends UI into confusing state

mtlynch, 2021-09-14:
We added an X button to all overlays in #782, but we forgot to disable it in situations where the user shouldn't be able to close the overlay.

In situations where we're hiding a "close" button during some asynchronous process, we should also disable the X button so that the user can't close the dialog to reach a confusing state.

https://user-images.githubusercontent.com/7783288/133298198-3da9daba-6b6a-4483-b847-521b1620c3c9.mp4

jotaen, 2021-09-23:
We also had the idea of a unified “progress” state for the overlays, as demonstrated in https://github.com/tiny-pilot/tinypilot/issues/714. As far as I can see, the problem with the `X` button only appears with the progress state, so one could argue it’s a “conceptual” problem that’s better solved generally.

I just wanted to mention this again “for the records”. Considering that the problem itself is confined, it might make sense to just solve it directly instead of investing more time into a generic solution.

mtlynch, 2021-09-23:
Yeah, for now I just want a narrowly-scoped solution, but it's good to remember that this is a problem we'll solve when we move to the more reusable progress states.