tiny-pilot/tinypilot #807: Offer option to redact sensitive information before sharing logs

mtlynch, 2021-11-01:
### Problem

There are certain log messages that can contain sensitive information, and we want to protect users from accidentally leaking these details before sharing their logs.

Example:

* If we fail to execute a keystroke, the keystroke (or sequence of keystrokes) may contain sensitive data that the user doesn't want to share in their logs if they post for debugging help.
* On TinyPilot Pro, users might want usernames to appear in the TinyPilot logs (e.g. `user123 logged in`, `user123 signed out`), but they'd leak sensitive information if they shared those logs.

### Proposal

We add a special annotation for log calls that can log sensitive information like `logger.info("[SENSITIVE] User %s logged in from %s", username, ip_address)`.

And then in the debug logs UI, we make a more distinct "Sharing" section and add a checkbox (off by default) labeled "Include sensitive data." When the checkbox is unchecked, we do a regex at the JS level to strip out any log lines that include the string `[SENSITIVE]`.

[image]

jotaen4tinypilot, 2021-11-09:
I quickly wanted to share an alternative UI approach, just for discussion. It’s just an idea, so I think the checkbox suggestion above would work just fine.

[image]

- Instead of a checkbox, we could use a toggle button. When enabled, it instantly applies the obfuscation to the displayed logs. That way, the user can review how the sanitized version of the logs look. Without this, the user couldn’t really be sure which parts exactly are hidden, so it may leave uncertainty.
- When clicking one of the two blue buttons, the current state of the toggle button is respected. So the notion is: “however it’s shown below is how it’s shared.”
- When generating a shareable URL with “hiding” turned on, we could show an additional message, just for re-assuring the user that their setting was taken over.

[image]

mtlynch, 2021-11-09:
Oh yeah, I like that much better. I agree it's better for the user to be able to see exactly what's going to be uploaded.

jotaen4tinypilot, 2021-11-15:
@mtlynch Regarding the backend, I just wanted to double-check whether you happen to have an idea in mind for adding such a `[SENSITIVE]` marker in a neat (native) way? The `logging` package provides some room for customization, but the only straightforward way I could find was registering separate methods for every log level, like `logger.info_sensitive(...)`, which seems overkill to me.

My current thought is to just provide a static helper method and do like `logger.info(log.sensitive('User %s just signed in'), user.name)`. That way the logic is encapsulated on the one hand, and the implementation is still reasonably straightforward.

mtlynch, 2021-11-15:
Either way works for me. I have a slight preference for `logger.info_sensitive` since it leads to more concise logging and three new methods doesn't sound that crazy, but both approaches feel about equal.

jotaen4tinypilot, 2021-11-17:
Hm, maybe dedicated log levels are not too bad after all, they look a bit nicer indeed. I’ll give that a try then.

jotaen4tinypilot, 2021-11-26:
Let’s see where we are – we have 3 options on the table right now:

1. `logger.info(log.sensitive('Some message %s'), x)`
2. https://github.com/tiny-pilot/tinypilot/pull/830
3. https://github.com/tiny-pilot/tinypilot/pull/832

My favourite is 2, because it’s the nicest to use. We need to buy it with some complexity, however, and I’m unsure about that, to be honest. If we think that wouldn’t be worth it, I would vote for 1, because that’s most straightforward while being (in contrast to 3) not hacky.

mtlynch, 2021-11-30:
I like option (2) as well. The complexity isn't too bad. It felt worse when we were fighting with our linters, but now that we've found smooth integrations with the linters, I think (2) works. If it causes problems, it's pretty easy to back out and replace with (1) if we need to.

jotaen4tinypilot, 2021-12-02:
Turns out, there are no log statements left to be converted in Pro actually. (Which I could have checked before making you merge over the latest changes… 🙈 )

But this also poses the question: shall we add some, e.g. in [`auth.py`](https://github.com/tiny-pilot/tinypilot-pro/blob/master/app/auth.py)?

mtlynch, 2021-12-02:
Nope, we can close this now. Adding the log messages to Pro is a separate bug (of which you're now the proud new owner 
🙂).

jotaen4tinypilot, 2021-12-02:
Haha, alright, that clears that up then.

jotaen4tinypilot, 2021-12-02:
Actually, let me re-open until https://github.com/tiny-pilot/tinypilot/pull/813 is through.

closed 2021-12-03 by jotaen4tinypilot