tiny-pilot/tinypilot #808: Add NumLock to on-screen keyboard

mtlynch, 2021-11-03:
The TinyPilot on-screen keyboard has no NumLock. This can lead to a case where the NumLock on the user's keyboard is out of state with the remote system:

https://forum.tinypilotkvm.com/-172/numlock-and-ten-key-numbers-issue

### Workaround

In the meantime, the user can work around this by pasting the following snippet in their JavaScript console:

```javascript
let key = document.getElementById("on-screen-keyboard").shadowRoot.querySelector('keyboard-key[code="ScrollLock"]');
key.innerText = "Num Lock"
key.code = "NumLock"
```

This temporarily converts the Scroll Lock key on the on-screen keyboard to a Num Lock key.

gazhay, 2025-09-24:
This doesn't appear to work anymore (the workaround)
Numlock key appears but certainly the windows remote machine doesn't respond to it

gazhay, 2025-09-24:
Also I use
```
javascript: (() => { document.querySelector('remote-screen').style.setProperty("--status-bar-height","300px")})();
```
as a bookmarklet to get the usable view out of the way of the keyboard.

gazhay, 2025-09-25:
The correct code to run in 2025 browsers (chrome at least)

```
let key = document.getElementById("on-screen-keyboard").shadowRoot.querySelector('keyboard-key[code="ScrollLock"]');
key.innerText = "Num Lock"
key.setAttribute("code","NumLock")
```

Or, if you want to add the key to a spare slot
```
let key = document.getElementById("on-screen-keyboard").shadowRoot.querySelector('keyboard-key.hidden:nth-child(2)');
key.innerText = "Num Lock"
key.setAttribute("code","NumLock")
key.classList.remove("hidden")
```