tiny-pilot/tinypilot #818: TinyPilot misinterprets AltGr on fr-ca keyboards

mtlynch, 2021-11-16:
A user reported that the AltGr key is not working correctly on a French Canadian keyboard + target (fr-ca).

I haven't been able to reproduce it exactly, but I do notice that TinyPilot fails to forward the keystrokes accurately. I tried sending AltGr + E, which is supposed to send `€` on a fr-ca keyboard, but the remote machine sees `e`.

* [What the browser sends](https://gist.github.com/mtlynch/8baafea9458862d45165a666be36ef68#file-raw-js-md)
* [What TinyPilot sees](https://gist.github.com/mtlynch/8baafea9458862d45165a666be36ef68#file-tinypilot-simplified-md)

The key point is on the keystroke. The `keydown` event looks like this:

```javascript
{
    "code": "KeyE",
    "key": "€",
    "altKey": false,
    "ctrlKey": false,
    "metaKey": false,
    "shiftKey": false,
    "location": 0
}
```

But the TinyPilot-generated event looks like this:

```python
{'code': 'KeyE', 'ctrlLeft': True, 'key': '€'}
```

So TinyPilot is sending `ctrlLeft: True` even though the key event said `"ctrlKey": false`.

This is something that [`_fixInternalModifierStates`](https://github.com/tiny-pilot/tinypilot/blob/422463dd86ea58bb2bcd51e5fc0f5eee60509094/app/static/js/keyboardstate.js#L59) is supposed to handle, and the code looks correct, but we need to debug further because something is going wrong.

### Testing note

To test this, you need to change both the client and target machine to fr-ca.

* Ubuntu: Settings > Keyboard > Input Sources -> Add French (Canada) and move it to the top of the list.
* Win10: Start > Language Settings > Add a Language, then select it from the system tray

mtlynch, 2021-11-18:
It's very likely to be a bug in `_fixInternalModifierStates`. The user reporting the issue rolled back to 2.2.2 (before we added in that code), and now it's working again.

https://forum.tinypilotkvm.com/-150#post-10

jdeanwallace, 2022-02-03:
# Update

I'm struggling to recreate this issue. 
1. @mtlynch you mentioned that pressing `AltGr + E`, should produce the `€` character on a `fr-ca` keyboard layout. However, that doesn't match the [`fr-ca` keyboards shown online](http://kbdlayout.info/KBDFC/) 🤔 
2. The [user mentioned](https://forum.tinypilotkvm.com/-150#post-5) that they using the following setup:
    > running windows, using a french canadian physical keyboard, french canadian layout
    
    I don't have a windows machine on hand, so maybe that's why I can't reproduce the original issue? 🤔 
3. Am I correct in saying that the original issue that the user is facing stems from this 👇 
    > pressing `AltGr` and seeing `[Control] [AltGraph]` on the bottom bar of the TinyPilot window (even if I never pressed `Control`)
4. I found [this mention](https://developer.mozilla.org/en-US/docs/Web/API/KeyboardEvent/getModifierState#modifier_keys_on_gecko) of windows systems equating `Alt + Ctrl` to `AltGr`.
5. I found [this random microsoft support issue](https://answers.microsoft.com/en-us/microsoftedge/forum/all/pressing-altgraph-also-presses-ctrl/87191eec-fcc0-4f0c-809b-382e8af8a0ce), confirming that there's a built-in feature with the German (QWERTZ) keyboard, where pressing the `AltGr` key acts as if `Alt + Ctrl` keys are pressed together. This suggests that each keyboard might have it's own way interpreting the `AltGr` key. However, it's more likely this is just how windows handles `AltGr` presses 🤔 
6. I can confirm that pressing `AltGr + 2` while using `fr-ca` keyboard layouts everywhere, produces the following:
    * Using target machine (Ubuntu) in text editor: `@`
    * Using source machine (MacOS) in text editor: `@`
    * Using source machine (MacOS) in TinyPilot web-UI:
        * Produces `@` onto the target machine
        * Shows `[Alt] [@]` on bottom bar
7. Interestingly, pressing `AltGr + Ctrl + 2` while using a `fr-ca` keyboard layout on the source machine (MacOS) in TinyPilot web-UI, shows `[Alt] [Control] [Unidentified]` on the bottom bar. Producing the following JS keyboard event:
    ```javascript
    {
      altKey: true,
      code: "Digit2",
      ctrlKey: true,
      key: "Unidentified",
      metaKey: false,
      shiftKey: false
    }
    ```
8. Some bonus setup weirdness that I found through pressing different keyboard buttons:
    * "Canadian French - CSA" on MacOS is `fr-CA` layout: http://kbdlayout.info/KBDFC/
    * "French (Canada)" on Ubuntu is `en-CA` layout: http://kbdlayout.info/KBDCA/
    * "French (Canada, legacy) on Ubuntu is `fr-CA` layout: http://kbdlayout.info/KBDFC/

mtlynch, 2022-02-03:
@jdeanwallace - Quick suggestion

Can you try writing a unit test for `keyboardstate.js` that passes in these keystrokes to `onKeyDown`:

```javascript
{
    "code": "ControlLeft",
    "key": "Control",
    "altKey": false,
    "ctrlKey": true,
    "metaKey": false,
    "shiftKey": false,
    "location": 1
},
{
    "code": "",
    "key": "Alt",
    "altKey": true,
    "ctrlKey": true,
    "metaKey": false,
    "shiftKey": false,
    "location": 0
},
{
    "code": "KeyE",
    "key": "€",
    "altKey": false,
    "ctrlKey": false,
    "metaKey": false,
    "shiftKey": false,
    "location": 0
}
```

And then does something like:

```javascript
assertFalse(keyboardState.isKeyPressed("ControlLeft"));
```

That was the part of the bug that felt most tangible to me. Even though I couldn't reproduce what the customer was seeing, I could see TinyPilot not maintaining key state correctly, and I suspect that's what's causing the issue. If the unit test fails, then it confirms there's a bug in the keyboard state handler.

closed 2022-02-07 by jdeanwallace