tiny-pilot/tinypilot #821: Offer a mechanism for press-and-hold a keystroke

mtlynch, 2021-11-22:
When TinyPilot's backend receives a keystroke, it passes the keystroke and then sends an HID signal to [release all keys](https://github.com/tiny-pilot/tinypilot/blob/d444c0538a242a3a083c8745656d2f21b3f3c59a/app/hid/keyboard.py#L10L13).

The original intent was to prevent TinyPilot from thinking that keys are "held" when there's actually just latency on the network connection. It's possible that we can delete the `release_keys` logic and just wait for the client to send the signal for key release (`keyup` from the browser).

In any case, we should offer a way for users to press and hold a key, either by getting rid of the forced `release_keys` call or by adding an option to the on-screen keyboard to press and hold keys until they're released.

Steroid-Factory, 2022-07-01:
Do you have a specific example of where and how this feature would be used?
Would this feature affect both control_keys and hid_keycode on key down?

mtlynch, 2022-07-01:
@Steroid-Factory - Thanks for checking it out!

Control keys currently have special behavior where they stay pressed until the user explicitly releases them. The idea is to change behavior so that all keys behave that way. Although my concern is that on connections with high latency, it will cause keys to repeat because maybe a user only presses the A key for 20 ms, but the "release" event takes 400ms to reach the backend, so there would be unexpected behavior.

The first step would be to test out what it's like if regular keys release like control keys under high latency. If it works fine, we just make the change. If it does, maybe we add intelligence so that if RTT for keystroke events is > 100ms, then we preserve the current behavior, and otherwise we wait for the user to explicitly release the key.