tiny-pilot/tinypilot #822: Distorted screen on BIOS

yateya, 2021-11-23:
## Description

I have some devices like this [SF110 H470](https://www.ecs.com.tw/en/Product/System/SF110_H470/overview). The manafuture logo appears correctly, and but when I enter BIOS, the screen is distorted as shown.

## What's the behavior that you expect?

video appears normally.

## What's happening instead?

video is distorted.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Click on Del button to enter BIOS
4. See error

## Screenshots

[image]

https://user-images.githubusercontent.com/7525948/142968706-921fbceb-ff72-402e-aa49-ddaa45d7cf9b.mp4

## Logs

http://logs.tinypilotkvm.com/v5hXyn1K

## Trials

I tried to [change resolution](https://github.com/tiny-pilot/tinypilot/issues/786), [change edid](https://github.com/tiny-pilot/tinypilot/issues/344), [changing FPS](https://github.com/tiny-pilot/tinypilot/issues/670) and [NUC edid fix](https://github.com/tiny-pilot/tinypilot/issues/142). Still have same problem.

Thanks.

NiKiZe, 2021-11-23:
Looks like the BIOS has a bug where the text renders strange due to low resolution. Reason for saying this is that it only seems to be the text that is of.

mtlynch, 2021-11-30:
Thanks for reporting this, @yateya and for the detailed writeup.

I unfortunately don't have a solution, as it seems to be a strange interaction between the target computer's GPU and the TinyPilot HDMI capture. I'll keep this open so that if we find other instances, we can look for similarities and possible workarounds.