tiny-pilot/tinypilot #824: Set static IP from the web UI

mtlynch, 2021-11-23:
We've gotten several requests for letting the user specify a static IP from the web UI.

There's [a way to do it](https://forum.tinypilotkvm.com/-47/setting-static-ip-address) if you SSH in, but it's a bit difficult at it involves editing config files. It would be helpful if we could offer a way to do this through the web UI.

mtlynch, 2022-06-10:
We now have a convenience script for [setting a static IP](https://tinypilotkvm.com/faq/static-ip), but we're now getting even more requests for a way to do this in the web UI, so prioritizing this higher.

db39, 2023-08-22:
A [user on the TinyPilot forum](https://forum.tinypilotkvm.com/-742/three-critical-improvements-needed-for-the-tinypiloy-voyager-2a) has requested this web interface functionality to set their static IP instead of via the command line.