tiny-pilot/tinypilot #829: Is it possible to Wake the Computer without WOL? I tried to type on the keyboard, but didn't work

wilsonnkwan, 2021-11-25:
Hi, 

My target computer does not support wake on lan for some reason, so I was trying to pass keystrokes into the target system while it is sleeping, it doesn't seem to work. I suspect OTG input is not working when the target computer is asleep.
But interestingly, I can wake up the target computer with a physical keyboard or mouse.

Is there anyway that we can wake the computer up while it is asleep other than WOL?

Thank you.

Warmest Regards,
Wilson

mtlynch, 2021-11-30:
Unfortunately, I don't have a workaround for getting target machines to wake up if they fail to respond to TinyPilot's keystrokes while asleep.

Have you checked your BIOS settings? Usually, if it's not responding to WOL, you can enable it there, but I've seen some motherboards that claim to support WOL and they actually don't (or only certain BIOS revisions do).

closed 2021-11-30 by mtlynch

jlian, 2022-08-04:
There's a [kernel patch](https://github.com/raspberrypi/linux/issues/3977#issuecomment-1200368214) + small [`init-usb-gadget` driver](https://github.com/tiny-pilot/ansible-role-tinypilot/compare/master...jlian:ansible-role-tinypilot:master) updates that now fixes this. Works on my TinyPilot CE setup.

wilsonnkwan, 2022-08-04:
John, thanks for this but need some pointers, how to patch this? Not really
good at manual patching. Any help is deeply appreciated. Thanks

jlian, 2022-08-04:
I had to learn to patch and compile the Linux kernel for the first time too! It’s different depending on your setup, but I took some notes here https://github.com/jlian/linux-kernel-cross-compile

Links I referenced:
- https://akhmad.id/compile-linux-kernel-on-mac-m1/
- https://github.com/geerlingguy/raspberry-pi-pcie-devices/tree/master/extras/cross-compile
- https://www.raspberrypi.com/documentation/computers/linux_kernel.html

mtlynch, 2022-08-08:
@jlian - Thanks so much for taking the time to write up those notes and share those references!

wilsonnkwan, 2022-08-08:
@mtlynch , Possible to incorporate in future release? hahahaha
Sorry, total noob here, not sure how to compile this after reading the instructions.

mtlynch, 2022-08-08:
We'll likely incorporate it if Linux accepts the patch upstream, though it's going to be a while, as it's nontrivial to distribute new kernel patches.

luckman212, 2024-05-11:
In May 2024 is there any progress here? My target system just went to sleep and I can't wake it up via mouse/keyboard... argh

db39, 2024-05-13:
>is there any progress here?

It doesn't look like there has been any upstream changes / patches, unfortunately. So this issue still exists.

As a work around, we created a simple [mouse jiggler](https://tinypilotkvm.com/faq/mouse-jiggler) to keep target machines from entering sleep mode. Alternatively, you can edit your target machine's settings to prevent it from entering sleep mode.