tiny-pilot/tinypilot #831: Clean up stale branches?

jotaen4tinypilot, 2021-11-26:
We have a total of 314 branches right now, most of which [are stale](https://github.com/tiny-pilot/tinypilot/branches/stale). Can we delete them, or is there a reason for keeping them? Ditto for our other repos, though [in some it’s not as bad](https://github.com/tiny-pilot/tinypilot-pro).

[Here is one way of doing bulk-deleting](https://stackoverflow.com/a/69089905), which we could use. I’d be happy to take this on, it shouldn’t take long.

This is not a big deal at all, it just would make managing remote branches a bit easier, e.g. when QAing something or renaming a branch etc.

(That’s how it looks in Git Tower)
[image]

mtlynch, 2021-11-29:
Sure you can delete them. I'd like to preserve branches that were never merged into master, but it's not a big deal if we just delete everything stale if that's easier.

mtlynch, 2023-04-06:
Closing, as we're now down to 67 branches, and they're unmerged branches that I'd like to keep around for posterity in case they're linked from bugs.

closed 2023-04-06 by mtlynch