tiny-pilot/tinypilot #838: Automatically merge from Community -> Pro

mtlynch, 2021-11-30:
Background: https://github.com/tiny-pilot/tinypilot-pro/issues/280

We should create a Github Action that keeps TinyPilot Pro in sync with TinyPilot Community. When someone makes a PR into the tiny-pilot/tinypilot, the Github Action should generate a PR of the same change into the tiny-pilot/tinypilot-pro repo and assign it to the author for review.

If we get an outside contribution from someone who doesn't have access to Github Pro, it would be good if the PR can go to the person who approved the PR (or default to me), but it's also fine if we have to handle those cases manually.

I've added a Github secret to this repo called `PR_BOT_PAT`. It's the PAT for a bot user that has access to the tiny-pilot/tinypilot-pro repo.

mtlynch, 2021-12-02:
@rw4nn - I tested this out with #840 but it failed to get the branch:

```text
/usr/bin/git checkout --progress master
  Switched to branch 'master'
  Your branch and 'origin/master' have diverged,
  and have 558 and 1 different commits each, respectively.
    (use "git pull" to merge the remote branch into yours)
  /usr/bin/git reset --hard origin/master
  HEAD is now at 6d4ecc8 Merge pull request #295 from tiny-pilot/merge-community-17eee72
  /usr/bin/git -c protocol.version=2 fetch --no-tags --progress --no-recurse-submodules --unshallow origin community-changes-e3cbe2028cddec753849d83760a0d6014906f02b:refs/remotes/origin/community-changes-e3cbe2028cddec753849d83760a0d6014906f02b
  fatal: couldn't find remote ref community-changes-e3cbe2028cddec753849d83760a0d6014906f02b
  Pull request branch 'community-changes-e3cbe2028cddec753849d83760a0d6014906f02b' does not exist yet.
```

https://github.com/tiny-pilot/tinypilot/runs/4399899451?check_suite_focus=true

rw4nn, 2021-12-03:
Ok, same workflows permission issue:

```
/usr/bin/git push --force-with-lease origin HEAD:refs/heads/community-changes-e3cbe2028cddec753849d83760a0d6014906f02b
  To https://github.com/tiny-pilot/tinypilot-pro
   ! [remote rejected] HEAD -> community-changes-e3cbe2028cddec753849d83760a0d6014906f02b (refusing to allow a Personal Access Token to create or update workflow `.github/workflows/merge-pro.yml` without `workflow` scope)
  error: failed to push some refs to 'https://github.com/tiny-pilot/tinypilot-pro'
  Error: The process '/usr/bin/git' failed with exit code 1
```

I don't quite understand because I already merged the `merge-pro.yml` change (deleting it on pro).

Anyway, can you try giving the workflows permission to the bot? @mtlynch And we'll see what happens on the next merge, thanks!

PR #839 (2021-12-02): Automatically merge from Community -> Pro
On a new commit to master branch in community repo, a merge PR is created in the pro repo, assigned to the author of the triggering commit.

When the committer has no access to the pro repo, the PR is still created but with no assignee.

Resolves #838.

---
This change is [[image]](https://reviewable.io/reviews/tiny-pilot/tinypilot/839)