tiny-pilot/tinypilot #881: Wiki update

yeyeto2788, 2022-01-05:
Hey guys! 

Awesome work done on the project 🥳 . I just finished setting up my first device with tinypilot and looks promising 😃 .

I wanted to contribute with some minor issues I found on the documentation. Before I list them down let me first tell you about my setup.

**SBC:** Raspberry Pi Zero W
**HDMI capture:** C779 CSI-HDMI bridge module

Steps I followed:

1. Install everything as default following repo `README.md` (everything went smooth) you can see the output if needed on the [installation_001.log](https://github.com/tiny-pilot/tinypilot/files/7812848/installation_001.log) file.
2. Edit the `settings.yml` with the needed attributes/variables for the rpi 0 (**It failed**) the output is in the [installation_002.log](https://github.com/tiny-pilot/tinypilot/files/7812862/installation_002.log) file.
  * The attributes/variables were as in the [Pi Zero Optimizations](https://github.com/tiny-pilot/tinypilot/wiki/Pi-Zero-Optimizations) page:
    ```yaml
     ustreamer_desired_fps: 15
     ustreamer_drop_same_frames: 15
     ```
3. Edit once again the `settings.yml`with the `ustreamer_use_dv_timings: true` attribute/variable as I saw it on the issue [#93](https://github.com/tiny-pilot/tinypilot/issues/93#issuecomment-724285252) ending with a `setting.yaml` file like this:
   ```yaml
   ustreamer_capture_device: "tc358743"
   ustreamer_use_dv_timings: true
   ustreamer_desired_fps: 15
   ustreamer_drop_same_frames: 15
   ustreamer_port: 8001
   ustreamer_persistent: true
   ustreamer_encoder: omx
   ustreamer_format: uyvy
   ustreamer_resolution: 1920x720
   ```

Now to the files I visited that IMHO, have missing or wrong information:
- [Pi Zero Optimizations](https://github.com/tiny-pilot/tinypilot/wiki/Pi-Zero-Optimizations)
  - Would be great to add how to edit the `settings.yml` via command line as in other pages.
- [Example: TC358743 HDMI to CSI capture chip](https://github.com/tiny-pilot/tinypilot/wiki/Installation-Options#example-tc358743-hdmi-to-csi-capture-chip)
  - No reference to the `USE_TC358743_DEFAULTS` variable seen in the [quick-install](https://github.com/tiny-pilot/tinypilot/blob/master/quick-install#L57) script. (I know it's being set by reading the `settings.yaml` file but for some reason on the 3rd tried it work setting that variable before executing the script (Even with the script setting it up as empty on that line).
- [Installation Options](https://github.com/tiny-pilot/tinypilot/wiki/Installation-Options)
  - It references a `TINYPILOT_SETTINGS` variable which seems not correct as per the [quick-install](https://github.com/tiny-pilot/tinypilot/blob/master/quick-install#L42) looks for a **`TINYPILOT_SETTINGS_FILE`**
- [All pages above]()
  - I didn't find any information on how to create the user so I finally prior executing any of the commands proposed I needed to create the user and the group with
    ```console
    sudo useradd --create-home --system --user-group tinypilot
    ```
  - Maybe would be great to have a separate wiki page for editing the `settings.yml` file
- [Contributing](https://github.com/tiny-pilot/tinypilot/blob/master/CONTRIBUTING.md)
  - No section on how to edit the wiki or how to proceed or how to help on documenting.
  - No section regarding translation (I can help on the Spanish part)

P.D: 
- If any more information is needed, do ask me 😛 
- I can add these changes to the wiki if required. (Just need to know the procedure, like, if it is needed to activate the hooks when there is no code developed)

mtlynch, 2022-01-07:
Thanks for this!

Sorry for the delay on responding. I'm hoping to get to this next week.

yeyeto2788, 2022-01-10:
No problem. If you want me to do something, let me know.

mtlynch, 2022-01-19:
Sorry again for the delay!

>Would be great to add how to edit the settings.yml via command line as in other pages.

We could, but it's a little ugly to make sure we're doing it idempotently. If we run the same command 10 times, it shouldn't just add the same line, but it's a bit more complicated to implement it to prevent that.

>No reference to the USE_TC358743_DEFAULTS variable seen in the quick-install script. (I know it's being set by reading the settings.yaml file but for some reason on the 3rd tried it work setting that variable before executing the script (Even with the script setting it up as empty on that line).

The user can't set the `USE_TC358743_DEFAULTS ` variable, as it gets [set in the script itself](https://github.com/tiny-pilot/tinypilot/blob/f415705497e3ae5d785821c69aa739902f759a7c/quick-install#L57). What you can do is set add this line to `/home/tinypilot/settings.yml`:

```yaml
ustreamer_capture_device: tc358743
```

It didn't take effect before because the documentation had quotes around `tc358743`, but the grep is [not flexible on quotes](https://github.com/tiny-pilot/tinypilot/blob/f415705497e3ae5d785821c69aa739902f759a7c/quick-install#L59). I've updated the examples to get rid of the quotes.

>It references a TINYPILOT_SETTINGS variable which seems not correct as per the quick-install looks for a TINYPILOT_SETTINGS_FILE

The two variables are independent. Both only have meaning locally within the script/example.

>I didn't find any information on how to create the user so I finally prior executing any of the commands proposed I needed to create the user and the group with

Ah, you're right. That should have been a first step in the advanced installation instructions. I've added that to the [examples](https://github.com/tiny-pilot/tinypilot/wiki/Installation-Options).

>No section on how to edit the wiki or how to proceed or how to help on documenting.

Github doesn't have tools for letting us review wiki changes effectively, so it's basically just that anyone's allowed to edit, and everyone does their best to keep it accurate and clear.

>No section regarding translation (I can help on the Spanish part)

We unfortunately don't have resources to support multi-language documentation or UI, but if you publish a Spanish-language tutorial, I'll link to it.