tiny-pilot/tinypilot #888: uStream compilation error

yeyeto2788, 2022-01-13:
## uStreamer fails while compiling on 32-bit ARMv6
Briefly summarize the problem you're experiencing.

## What's the behavior that you expect?

*Compilation goes smoothly and my device outputs signal from HDMI* 😋 

## What's happening instead?

*Ansible reports that failed while compiling the software*

## What are the steps to reproduce this behavior?

1. Create `tinypilot` user by:
    `sudo useradd --create-home --system --user-group tinypilot`
2. Create `/home/tinypilot/setting.yml` with the following information:
    ```yaml
    ustreamer_capture_device: "tc358743"
    ustreamer_use_dv_timings: true
    ustreamer_desired_fps: 15
    ustreamer_drop_same_frames: 15
    ustreamer_port: 8001
    ustreamer_persistent: true
    ustreamer_encoder: omx
    ustreamer_format: uyvy
    ustreamer_resolution: 1920x720
    ```
3. Change ownership of the file:
    ```console
    sudo chown tinypilot:tinypilot "${TINYPILOT_SETTINGS_FILE}"
    ```
4. Set some variable before install
    ```console
    USE_TC358743_DEFAULTS=y
    TINYPILOT_SETTINGS_FILE="/home/tinypilot/settings.yml"
    ```
5. Install super tinypilot:
    ```console
    curl --silent --show-error https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | bash -
    ```
4. See error

## Screenshots

*Raspberry pi OS lite configuration*
[image]

## Logs
Line 437 of the log file below shows the error 😢 
[installation_001.log](https://github.com/tiny-pilot/tinypilot/files/7861768/installation_001.log)

yeyeto2788, 2022-01-13:
Hey guys! This issue seems to be related to [this](https://github.com/pikvm/ustreamer/issues/123)

yeyeto2788, 2022-01-13:
Confirmed, seems like the issue in on Debian Bullseye as I tried it with Buster and the compilation did go smoothly, here you can see the output of the test:
[installation_002.log](https://github.com/tiny-pilot/tinypilot/files/7862754/installation_002.log)

After reboot I get a different error (still no output 😢 :
```
TinyPilot log dump
https://tinypilotkvm.com
Timestamp: 2022-01-13T13:38:16+01:00

Software versions
TinyPilot version: 1.6.2-2-g8c924e8 8c924e8
uStreamer version: v3.26 a97f08e
OS version: Linux tinypilot 5.10.63+ #1496 Wed Dec 1 15:57:05 GMT 2021 armv6l GNU/Linux

TinyPilot state
temp=32.6'C
throttled=0x0

voltage logs

TinyPilot configuration
[Unit]
Description=TinyPilot - RPi-based virtual KVM
After=syslog.target network.target
StartLimitIntervalSec=0

[Service]
Type=simple
User=tinypilot
WorkingDirectory=/opt/tinypilot
ExecStart=/opt/tinypilot/venv/bin/python app/main.py
Environment=HOST=127.0.0.1
Environment=PORT=8000
Environment=APP_SETTINGS_FILE=/opt/tinypilot/app_settings.cfg
Restart=always

[Install]
WantedBy=multi-user.target

TinyPilot logs
-- Logs begin at Thu 2022-01-13 13:33:26 CET, end at Thu 2022-01-13 13:38:18 CET. --
Jan 13 13:33:52 tinypilot systemd[1]: Started TinyPilot - RPi-based virtual KVM.
Jan 13 13:34:30 tinypilot python[394]: 2022-01-13 13:34:30.571 __main__        INFO Starting app
Jan 13 13:38:00 tinypilot python[394]: 2022-01-13 13:38:00.540 socket_api      INFO Client connected
Jan 13 13:38:04 tinypilot python[394]: 2022-01-13 13:38:04.944 socket_api      INFO Client connected
Jan 13 13:38:06 tinypilot python[394]: Process ProcessWithResult-1:
Jan 13 13:38:06 tinypilot python[394]: Traceback (most recent call last):
Jan 13 13:38:06 tinypilot python[394]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 13 13:38:06 tinypilot python[394]:     self.run()
Jan 13 13:38:06 tinypilot python[394]:   File "/opt/tinypilot/app/hid/write.py", line 45, in run
Jan 13 13:38:06 tinypilot python[394]:     result.return_value = self._target(*self._args, **self._kwargs)
Jan 13 13:38:06 tinypilot python[394]:   File "/opt/tinypilot/app/hid/write.py", line 64, in _write_to_hid_interface_immediately
Jan 13 13:38:06 tinypilot python[394]:     with open(hid_path, 'ab+') as hid_handle:
Jan 13 13:38:06 tinypilot python[394]: PermissionError: [Errno 13] Permission denied: '/dev/hidg0'
Jan 13 13:38:07 tinypilot python[394]: 2022-01-13 13:38:07.003 socket_api      ERROR [SENSITIVE] Failed to write key: Control (keycode=ControlLeft). Failed to write to HID interface: /dev/hidg0. Is USB cable connected? [/SENSITIVE]
Jan 13 13:38:07 tinypilot python[394]: Process ProcessWithResult-2:
Jan 13 13:38:07 tinypilot python[394]: Traceback (most recent call last):
Jan 13 13:38:07 tinypilot python[394]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 13 13:38:07 tinypilot python[394]:     self.run()
Jan 13 13:38:07 tinypilot python[394]:   File "/opt/tinypilot/app/hid/write.py", line 45, in run
Jan 13 13:38:07 tinypilot python[394]:     result.return_value = self._target(*self._args, **self._kwargs)
Jan 13 13:38:07 tinypilot python[394]:   File "/opt/tinypilot/app/hid/write.py", line 64, in _write_to_hid_interface_immediately
Jan 13 13:38:07 tinypilot python[394]:     with open(hid_path, 'ab+') as hid_handle:
Jan 13 13:38:07 tinypilot python[394]: PermissionError: [Errno 13] Permission denied: '/dev/hidg0'
Jan 13 13:38:07 tinypilot python[394]: 2022-01-13 13:38:07.157 socket_api      ERROR [SENSITIVE] Failed to write key: Shift (keycode=ShiftLeft). Failed to write to HID interface: /dev/hidg0. Is USB cable connected? [/SENSITIVE]
Jan 13 13:38:07 tinypilot python[394]: Process ProcessWithResult-3:
Jan 13 13:38:07 tinypilot python[394]: Traceback (most recent call last):
Jan 13 13:38:07 tinypilot python[394]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 13 13:38:07 tinypilot python[394]:     self.run()
Jan 13 13:38:07 tinypilot python[394]:   File "/opt/tinypilot/app/hid/write.py", line 45, in run
[... 135 lines cut ...]
192.168.0.23 - - [13/Jan/2022:13:32:13 +0100] "GET /img/logo.svg HTTP/1.1" 200 2936 "http://tinypilot.local/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:95.0) Gecko/20100101 Firefox/95.0" "-"
192.168.0.23 - - [13/Jan/2022:13:32:13 +0100] "GET /third-party/fonts/overpass-mono-light.woff2 HTTP/1.1" 200 50112 "http://tinypilot.local/third-party/fonts/fonts.css" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:95.0) Gecko/20100101 Firefox/95.0" "-"
192.168.0.23 - - [13/Jan/2022:13:32:13 +0100] "GET /third-party/fonts/overpass-semibold.woff2 HTTP/1.1" 200 25032 "http://tinypilot.local/third-party/fonts/fonts.css" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:95.0) Gecko/20100101 Firefox/95.0" "-"
192.168.0.23 - - [13/Jan/2022:13:32:13 +0100] "GET /css/toggle.css HTTP/1.1" 200 1005 "http://tinypilot.local/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:95.0) Gecko/20100101 Firefox/95.0" "-"
192.168.0.23 - - [13/Jan/2022:13:32:13 +0100] "GET /socket.io/?EIO=4&transport=polling&t=NvJR9dh HTTP/1.1" 200 96 "http://tinypilot.local/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:95.0) Gecko/20100101 Firefox/95.0" "-"
192.168.0.23 - - [13/Jan/2022:13:32:13 +0100] "POST /socket.io/?EIO=4&transport=polling&t=NvJR9f8&sid=PmgbFbUI-iUtCHQBAAAA HTTP/1.1" 200 2 "http://tinypilot.local/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:95.0) Gecko/20100101 Firefox/95.0" "-"
[... cut ...]
192.168.0.23 - - [13/Jan/2022:13:33:18 +0100] "GET /stream?advance_headers=1 HTTP/1.1" 200 919929 "http://tinypilot.local/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:95.0) Gecko/20100101 Firefox/95.0" "-"
192.168.0.23 - - [13/Jan/2022:13:33:19 +0100] "GET /socket.io/?EIO=4&transport=websocket&sid=PmgbFbUI-iUtCHQBAAAA HTTP/1.1" 101 186 "-" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:95.0) Gecko/20100101 Firefox/95.0" "-"
```

**NOTE:** 
* Play special attention to the handler  `load TC358743 EDID file` from the `ansible-role-ustreamer`
* We may want to add this into the wiki as using OMX in Bullseye will probably fail.

mtlynch, 2022-01-13:
Thanks for reporting this, @yeyeto2788! This is a known issue, but uStreamer doesn't have an active issue for it, so I'll keep this one open. The uStreamer maintainer is working on a fix in the [m2m branch](https://github.com/pikvm/ustreamer/tree/m2m).

ThomasVon2021, 2022-03-20:
A similar situation was encountered.
ERROR! We were unable to read either as JSON nor YAML, these are the errors we got from each:
JSON: Expecting value: line 1 column 1 (char 0)

Syntax Error while loading YAML.
  mapping values are not allowed here

The error appears to be in '/var/tmp/tmp.rX0fipeWKZ.yml': line 5, column 15, but may
be elsewhere in the file depending on the exact syntax problem.

The offending line appears to be:

  sudo chown tinypilot:tinypilot "${TINYPILOT_SETTINGS}"
ustreamer_port: 8001
              ^ here

mtlynch, 2022-03-21:
@ThomasVon2021 - Can you share the exact command you're running that leads to this error?