tiny-pilot/tinypilot #892: Failed to forward mouse event: Failed to write to HID interface

void4main, 2022-01-18:
Hi all, not sure what's wrong or how to debug this:

## Description

Keyboard and mouse do not work.

Both devices are present
/dev/hidg0
/dev/hidg1

USB-A cable male-male connects the Tinypilot (rpi4) either to a Raspberry Pi 3 or Jetson Nano.
Both do not work.
I have no problems with the webinterface nor video. 
Powersupply is 3A and no errors in the logs.

## Logs

`Jan 18 20:32:22 pi03 python[523]: 2022-01-18 20:32:22.109 git             INFO Remote HEAD commit ID: f415705497e3ae5d785821c69aa739902f759a7c
Jan 18 20:32:24 pi03 python[3759]: Process ProcessWithResult-102:
Jan 18 20:32:24 pi03 python[3759]: Traceback (most recent call last):
Jan 18 20:32:24 pi03 python[3759]:   File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
Jan 18 20:32:24 pi03 python[3759]:     self.run()
Jan 18 20:32:24 pi03 python[3759]:   File "/opt/tinypilot/app/hid/write.py", line 45, in run
Jan 18 20:32:24 pi03 python[3759]:     result.return_value = self._target(*self._args, **self._kwargs)
Jan 18 20:32:24 pi03 python[3759]:   File "/opt/tinypilot/app/hid/write.py", line 65, in _write_to_hid_interface_immediately
Jan 18 20:32:24 pi03 python[3759]:     hid_handle.write(bytearray(buffer))
Jan 18 20:32:24 pi03 python[3759]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
Jan 18 20:32:24 pi03 python[523]: 2022-01-18 20:32:24.203 socket_api      ERROR [SENSITIVE] Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected? [/SENSITIVE]
Jan 18 20:32:26 pi03 python[3760]: Process ProcessWithResult-103:
Jan 18 20:32:26 pi03 python[3760]: Traceback (most recent call last):
Jan 18 20:32:26 pi03 python[3760]:   File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
Jan 18 20:32:26 pi03 python[3760]:     self.run()
Jan 18 20:32:26 pi03 python[3760]:   File "/opt/tinypilot/app/hid/write.py", line 45, in run
Jan 18 20:32:26 pi03 python[3760]:     result.return_value = self._target(*self._args, **self._kwargs)
Jan 18 20:32:26 pi03 python[3760]:   File "/opt/tinypilot/app/hid/write.py", line 65, in _write_to_hid_interface_immediately
Jan 18 20:32:26 pi03 python[3760]:     hid_handle.write(bytearray(buffer))
Jan 18 20:32:26 pi03 python[3760]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
Jan 18 20:32:26 pi03 python[523]: 2022-01-18 20:32:26.206 socket_api      ERROR [SENSITIVE] Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected? [/SENSITIVE]
Jan 18 20:32:34 pi03 sudo[3774]: tinypilot : PWD=/opt/tinypilot ; USER=root ; COMMAND=/opt/tinypilot-privileged/collect-debug-logs -q
Jan 18 20:32:34 pi03 sudo[3774]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=998)`

mtlynch, 2022-01-18:
Thanks for reporting this!

Are you connecting to the Pi's USB-C port or one of the USB-A ports?

void4main, 2022-01-18:
Thank you!

Only the USB-A ports.

mtlynch, 2022-01-18:
Oh, the USB-C port is the only one capable of emulating USB devices, so you need to connect to that port. See this tutorial for details:

https://tinypilotkvm.com/blog/build-a-kvm-over-ip-under-100#how-to-build-your-own-tinypilot

closed 2022-01-18 by mtlynch