tiny-pilot/tinypilot #906: Rpi4 virtual usb and mouse not working

anixon604, 2022-02-08:
## Description

Unable to use Virtual Keyboard and mouse on Rpi4

## What's the behavior that you expect?

That the usb and keyboard will work

## What's happening instead?

Nothing happens, see log below.

## What are the steps to reproduce this behavior?

1. Pi OS 5.10.92-v8+
2. run the basic quickstart script
4. See error

## Screenshots

*If applicable, add screenshots to help explain your problem.*

## Logs

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`
http://logs.tinypilotkvm.com/8Xg6nBi7

mtlynch, 2022-02-08:
Thanks for reporting this!

Could you share a few more details?

* What kind of computer are you connecting the Pi to?
* How are you connecting the Pi4 to the target computer?

anixon604, 2022-02-08:
I'm connecting it to a debian based Unraid server. I've tried USB-A (server) to USB-A (rpi) and am now currently using USB-A (server) and USB-C (rpi) as recommended.

I did a dmesg on the server and noticed this...

```
[98742.527088] usb 1-1: new high-speed USB device number 25 using xhci_hcd
[98747.943059] usb 1-1: device descriptor read/64, error -110
[98763.814946] usb 1-1: device descriptor read/64, error -110
[98763.916959] usb usb1-port1: attempt power cycle
[98764.544779] usb 1-1: new high-speed USB device number 26 using xhci_hcd
[98769.844911] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[98775.476867] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[98775.684837] usb 1-1: device not accepting address 26, error -62
[98775.799841] usb 1-1: new high-speed USB device number 27 using xhci_hcd
[98781.108834] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[98786.740786] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[98786.948750] usb 1-1: device not accepting address 27, error -62
[98786.948782] usb usb1-port1: unable to enumerate USB device
[99217.430059] usb 1-1: new high-speed USB device number 28 using xhci_hcd
[99222.563028] usb 1-1: device descriptor read/64, error -110
[99238.434904] usb 1-1: device descriptor read/64, error -110
[99238.650882] usb 1-1: new high-speed USB device number 29 using xhci_hcd
[99244.066858] usb 1-1: device descriptor read/64, error -110
[99259.938729] usb 1-1: device descriptor read/64, error -110
[99260.040741] usb usb1-port1: attempt power cycle
```

Could this be the RPI? If so, do you know how to resolve this on debian?

mtlynch, 2022-02-08:
USB-A to USB-A definitely won't work.

I'm not sure why you're still seeing errors after going USB-C to USB-A. Can you try power cycling both the server and the Pi 4?

anixon604, 2022-02-09:
```
[   38.684239] usb 1-1: device descriptor read/64, error -110
[   54.556240] usb 1-1: device descriptor read/64, error -110
[   54.666286] usb usb1-port1: attempt power cycle
[   55.294205] usb 1-1: new high-speed USB device number 4 using xhci_hcd
[   60.586272] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   66.218257] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   66.426207] usb 1-1: device not accepting address 4, error -62
[   66.543207] usb 1-1: new high-speed USB device number 5 using xhci_hcd
[   71.850274] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   77.482257] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   77.690204] usb 1-1: device not accepting address 5, error -62
[   77.693256] usb usb1-port1: unable to enumerate USB device
```
Still had the same problem...
```
root@defsig:~# lsusb
Bus 002 Device 002: ID 0bda:9210 Realtek Semiconductor Corp. RTL9210B-CG
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 007: ID 0d8c:0014 C-Media Electronics, Inc. USB Audio Device
Bus 001 Device 006: ID 0781:55a1 SanDisk Corp. Cruzer Spark
Bus 001 Device 009: ID 26ce:01a2 ASRock LED Controller
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Is the `Linux Foudation X root hub` the RPI? This is from my server..

anixon604, 2022-02-09:
I think we may be linked https://github.com/tiny-pilot/tinypilot/issues/908

mtlynch, 2022-02-10:
Oh, you're using the new 64-bit Raspbian:

>OS version: Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 **aarch64** GNU/Linux

We haven't tested that yet, so I think you're hitting a bug with the USB gadget driver. I've filed [a bug](https://github.com/tiny-pilot/ansible-role-tinypilot/issues/181) to track this.

TinyPilot will work on the 32-bit version of Raspbian buster or bullseye. If you're using an [HDMI to CSI bridge](https://github.com/tiny-pilot/tinypilot/wiki/HDMI-Capture-Devices#toshiba-tc358743-capture-devices), you'll need to use bullseye, but if you're just using a [USB dongle](https://github.com/tiny-pilot/tinypilot/wiki/HDMI-Capture-Devices#macrosilicon-ms2109-capture-devices), either buster or bullseye should work.