tiny-pilot/tinypilot #908: RPi Zero Usb Gadget init issue

maburz, 2022-02-09:
usb gadget driver fails to initialize properly on latest Raspberry PI OS Lite on Pi Zero board. There were no errors during TinyPilot installation.

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux

pi@raspberrypi:~ $ sudo ls -al /dev/h*
crw------- 1 root root 10, 183 Feb  8 21:53 /dev/hwrng

pi@raspberrypi:~ $ sudo journalctl -u usb-gadget
Feb 08 21:53:45 raspberrypi systemd[1]: Starting Initialize USB gadgets...
Feb 08 21:53:45 raspberrypi init-usb-gadget[304]: + set -u
Feb 08 21:53:45 raspberrypi init-usb-gadget[313]: +++ dirname /opt/tinypilot-privileged/init-usb-gadget
Feb 08 21:53:45 raspberrypi init-usb-gadget[312]: ++ cd /opt/tinypilot-privileged
Feb 08 21:53:46 raspberrypi init-usb-gadget[312]: ++ pwd
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + readonly SCRIPT_DIR=/opt/tinypilot-privileged
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + SCRIPT_DIR=/opt/tinypilot-privileged
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + source /opt/tinypilot-privileged/lib/usb-gadget.sh
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_DEVICE_DIR=g1
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_DEVICE_DIR=g1
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_GADGET_PATH=/sys/kernel/config/usb_gadget
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_GADGET_PATH=/sys/kernel/config/usb_gadget
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_DEVICE_PATH=/sys/kernel/config/usb_gadget/g1
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_DEVICE_PATH=/sys/kernel/config/usb_gadget/g1
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_FILE_CONFIG_PATH=/sys/kernel/config/usb_gadget/g1/functions/mass_storage.0/lun.0/file
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_FILE_CONFIG_PATH=/sys/kernel/config/usb_gadget/g1/functions/mass_storage.0/lun.0/file
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_CDROM_CONFIG_PATH=/sys/kernel/config/usb_gadget/g1/functions/mass_storage.0/lun.0/cdrom
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_CDROM_CONFIG_PATH=/sys/kernel/config/usb_gadget/g1/functions/mass_storage.0/lun.0/cdrom
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_STRINGS_DIR=strings/0x409
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_STRINGS_DIR=strings/0x409
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_KEYBOARD_FUNCTIONS_DIR=functions/hid.keyboard
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_KEYBOARD_FUNCTIONS_DIR=functions/hid.keyboard
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_MOUSE_FUNCTIONS_DIR=functions/hid.mouse
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_MOUSE_FUNCTIONS_DIR=functions/hid.mouse
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_MASS_STORAGE_FUNCTIONS_DIR=functions/mass_storage.0
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_MASS_STORAGE_FUNCTIONS_DIR=functions/mass_storage.0
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly USB_CONFIGS_DIR=configs
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_CONFIGS_DIR=configs
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly 'USB_ALL_CONFIGS_DIR=configs/*'
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_ALL_CONFIGS_DIR='configs/*'
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export readonly 'USB_ALL_FUNCTIONS_DIR=functions/*'
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ USB_ALL_FUNCTIONS_DIR='functions/*'
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: ++ export -f usb_gadget_activate
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + modprobe libcomposite
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + cd /sys/kernel/config/usb_gadget
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + mkdir -p g1
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + cd g1
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + echo 0x1d6b
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + echo 0x0104
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + echo 0x0100
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + echo 0x0200
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + mkdir -p strings/0x409
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + echo 6b65796d696d6570690
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + echo tinypilot
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + echo 'Multifunction USB Device'
Feb 08 21:53:46 raspberrypi init-usb-gadget[304]: + mkdir -p functions/hid.keyboard
Feb 08 21:53:47 raspberrypi init-usb-gadget[330]: mkdir: cannot create directory ‘functions/hid.keyboard’: No such file or directory
Feb 08 21:53:47 raspberrypi systemd[1]: usb-gadget.service: Main process exited, code=exited, status=1/FAILURE
Feb 08 21:53:47 raspberrypi systemd[1]: usb-gadget.service: Failed with result 'exit-code'.
Feb 08 21:53:47 raspberrypi systemd[1]: Failed to start Initialize USB gadgets.

maburz, 2022-02-09:
Some suspicious lines in dmesg:
[   29.249245] usb_f_hid: no symbol version for module_layout
[   29.249286] usb_f_hid: version magic '5.10.76-v7l+ SMP mod_unload modversions ARMv7 p2v8 ' should be '5.10.92+ mod_unload modversions ARMv6 p2v8 '

mtlynch, 2022-02-10:
Is it a Pi Zero or a Pi Zero W? I don't think Pi Zero has USB OTG functionality, which TinyPilot needs to emulate USB devices.

maburz, 2022-02-11:
Its Pi Zero W, so it should have USB OTG capability..

mtlynch, 2022-02-11:
Oh, then you're running into this:

https://github.com/tiny-pilot/ansible-role-tinypilot/issues/181

We compiled a custom version of the USB HID gadget driver that has fixes not yet available in the Raspbian OS, but it's compiled for armv7l and we forgot to verify that the target system is the same architecture. It looks like Pi Zero W is armv6l.

As a workaround, you could add these lines to `/opt/tinypilot/quick-install` at [this spot](https://github.com/tiny-pilot/tinypilot/blob/1e1ddadf29849f2c809c3244f6054fc23a8afcbf/quick-install#L167) and then re-run it:

```bash
pushd "${TINYPILOT_ROLE_NAME}"
git checkout df94b266f6814f553181b5ca5761643127535f43
popd
```

Sorry about the inconvenience!

closed 2022-02-11 by mtlynch

maburz, 2022-02-11:
After adding those lines, when running install script again:
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'python-pip' is available"}

mtlynch, 2022-02-11:
Is this on Raspbian Bullseye or Buster?

We haven't tested as much on Bullseye yet, so compatibility will be better on Buster. If you're using an HDMI to CSI bridge, that will only work on Buster for the time being, anyway.