tiny-pilot/tinypilot #917: TinyPilot refuses to start if secret key is in invalid state

mtlynch, 2022-02-23:
Two users have now reported being unable to access their TinyPilot web dashboard because the `.flask-secret-key` was in an invalid state:

* https://forum.tinypilotkvm.com/-195
* https://forum.tinypilotkvm.com/-252

I suspect that this happens occasionally due to crashes or power cuts at unlucky moments - after the file is created but before we finish populating it.

I think that instead of creating the key file at `~/.flask-secret-key`, we should create it at `~/.tmp-flask-secret-key` and move it after we populate the data and set the permissions. This prevents the key from getting stuck in an illegal state because moves are atomic, so the file should either exist at `~/.flask-secret-key` in the correct state or not be there at all.

mtlynch, 2022-02-24:
We're getting a lot of reports of this all of a sudden and I'm not sure why.

@jotaen4tinypilot - Can you take this one? Maybe we can include it in 2.4.0 instead of 2.4.1.

jotaen4tinypilot, 2022-02-25:
I’ll look into it.

It seems odd to me, though, that such a random failure would occur repeatedly. I’d expect the [key creation procedure](https://github.com/tiny-pilot/tinypilot/blob/b8f0d71474db2742609fee86a7e6b78610c5ba63/app/secret_key.py#L76-L79) to take a few milliseconds at most. So I wonder whether there might be other causes that we are not aware of yet.

I think using (and extending) the atomic file is a good idea, because it eliminates corrupt in-between states. In addition to that, we could consider to recreate the secret key in case we detect the file or its contents as invalid. E.g. we catch [the `InvalidSecretKeyError` errors](https://github.com/tiny-pilot/tinypilot/blob/b8f0d71474db2742609fee86a7e6b78610c5ba63/app/secret_key.py#L54), and instead of letting the app crash, we try to remove and recreate the file “on the fly”. (Only if that keeps failing, we would abort.) I think the side-effects of that would be that the user needs to login again afterwards, because their session cookie is no longer accepted by flask.

mtlynch, 2022-02-25:
Yeah, I agree about recreating the key. When it was just two reports a few months apart, I thought two people just got unlucky, but now that there's been a third, it makes sense to recover more proactively if we hit this.

jotaen4tinypilot, 2022-02-25:
I look into the re-creation bit once https://github.com/tiny-pilot/tinypilot/pull/921 is in (the third PR).

closed 2022-03-02 by jotaen4tinypilot