tiny-pilot/tinypilot #928: Make decision whether `atomic_file.create` should overwrite existing file

jotaen4tinypilot, 2022-03-03:
`atomic_file.create` uses [`shutil.move`](https://github.com/tiny-pilot/tinypilot/blob/a0aeab2add6169f61ea7598253b537fe314bfd4a/app/atomic_file.py#L40) under the hood. If a file with the same name already exists at the target location, it seems like that is silently overwritten then.

This behaviour is neither specified nor tested right now, so I don’t think we have considered this case. It would help if the desired behaviour was defined, and then we should add tests and a comment (one way or the other).

I don’t have a strong preference, I think good points can be made either way.

In case we decide that overwriting is what we want, we should double check all calling code, to see whether we need to adjust or cleanup things there.

mtlynch, 2022-03-03:
Good catch! I think we should match the semantics of `open(path, 'w')`, so it should overwrite existing files.

jdeanwallace, 2022-03-04:
> I think we should match the semantics of open(path, 'w'), so it should overwrite existing files.

I agree with this.