tiny-pilot/tinypilot #929: usb-gadget service fails on 64-bit Raspberry Pi OS

yamato225, 2022-03-07:
## Description

Usb-gadget service fails on 64-bit Raspberry Pi OS.
same error which reported here. https://forum.tinypilotkvm.com/-220/usb-devices-not-created

## What's the behavior that you expect?

I suppose tinypilot can work on 64bit Raspberry Pi OS.

## What's happening instead?

After installation, the usb-gadget failed.
And I have confirmed that installation completed successfully on 32bit - OS

## What are the steps to reproduce this behavior?

1. Install Raspberry Pi OS 64bit
2. Install tinypilot following README.md
3. Reboot
4. usb-gadget service fails, the keyboard and mouse doesn't work.

mtlynch, 2022-03-07:
Thanks for reporting this! Only 32-bit is supported right now, but this it's good to have a bug to track this if we add 64-bit compatibility.

yamato225, 2022-03-07:
Thank you for reply. Oh, I understood.
Then, how's add a warning message into `quick-install` like this?
This will help people having same troulble.

```bash
if [[ $(uname -a | grep -c "aarch64") -gt 0 ]];then
    echo "Warning: currently tinypilot doesn't support on 64bit OS. Use 32bit or wait a moment."
    exit 0
fi
```

mtlynch, 2022-03-07:
That's a good idea. `uname -m` actually prints out just the architecture:

```bash
$ uname -m
armv7l
```

So maybe just compare the output of `uname -m` to `aarch64`. We'd also want to print to stderr and exit with a non-zero exit code.

thomstratton, 2022-04-16:
I am experiencing same problem, thought it might help if I posted this work-around script I am using on mine.  This just restarts the gadget without restarting the pi.

~~~bash
sudo /opt/tinypilot-privileged/remove-usb-gadget
sudo /opt/tinypilot-privileged/init-usb-gadget
~~~

mtlynch, 2022-04-18:
Thanks for sharing that, @thomstratton! A cleaner solution might be to restart the service, though I haven't yet tested on a 64-bit build:

```bash
sudo service usb-gadget restart
```

thomstratton, 2022-04-18:
aha!  Thanks that is better :)

billdenney, 2022-12-16:
I was just trying today on a 64-bit version, and was stopped by the installer check for 64-bit.  Is there a way that I could help track the crash with the 64-bit usb-gadget?  I'd like to help remove this limitation.

mtlynch, 2022-12-19:
@billdenney - Sure when it fails on 64-bit, it's exiting [at this point](https://github.com/tiny-pilot/tinypilot/blob/3592f8a2b7d475ac636ce3fa251eb088d9f25a6b/bundler/bundle/install#L43L49). If you attempt an install and then let it get to the failure, you should be able to edit the `install` script to remove those checks, then re-run `sudo ./install` to let it complete the install, and then you can debug the failure.

billdenney, 2022-12-19:
I have a successful installation!

Is there any specific way to put the usb-gadget through its paces or to induce a crash to help with debugging?

mtlynch, 2022-12-19:
Oh, that's good news. It could just be that whatever bug was preventing it from working was fixed upstream.

What version of Raspberry Pi OS did you install? Do you have the URL?

billdenney, 2022-12-19:
I installed the default version that was selected from the Raspberry Pi installer (as of about a month ago).  It's based on Debian 11 and I have updated all apt packages just before installation:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
```

Does that answer the question?

And, before did the usb-gadget service crash immediately or did it take a while?

I just got everything running about 5 minutes ago and my only testing has been to login and run ls on a text terminal.  I've not done significant testing, and I've not used the mouse.  (For the system that I'm connected to, I don't have any graphical software in use, so I can't easily test a mouse.)

mtlynch, 2022-12-19:
@billdenney - My understanding was that the service would fail to start at all on 64-bit Raspbian, so it looks like it's resolved.

closed 2022-12-19 by mtlynch