tiny-pilot/tinypilot #932: Tinypilot not displaying hdmi output from pc using vga to hdmi adaptor

taylor-schneider, 2022-03-14:
## Description

Tinypilot is showing the black "No signal" page. The vga-hdmi adaptor is working; I have verified it to work when connecting to a portabl 15" monitor and a 50" tv. Tinypilot is working with other adaptors.

## What's the behavior that you expect?

I expect to be able to see the output

## What's happening instead?

No output is shown

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Look at the screen

## Screenshots

Black No Signal

## Logs

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`

https://logs.tinypilotkvm.com/0zMaTThM

taylor-schneider, 2022-03-14:
The part of the logs that jumped out at me is:
```
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.530    stream] -- Device fd=10 opened
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.530    stream] -- Using input channel: 0
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.537    stream] -- Got new DV timings: resolution=1920x1080, pixclk=148500000
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.556    stream] -- Using resolution: 1920x1080
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.556    stream] -- Using pixelformat: UYVY
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.556    stream] -- Querying HW FPS changing is not supported
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.556    stream] -- Using IO method: MMAP
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.567    stream] -- Requested 5 device buffers, got 5
Mar 14 17:33:22 tinypilot ustreamer[698]: -- ERROR [2169484.567    stream] -- Unable to start capturing: Invalid argument
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.573    stream] -- Device fd=10 closed
Mar 14 17:33:22 tinypilot ustreamer[698]: -- INFO  [2169484.573    stream] -- Sleeping 1 seconds before new stream init ...
```

I started googling 'Unable to start capturing: Invalid argument' and found an article talking about reinstalling and updating the configs to specify a custom resolution: https://githubmemory.com/repo/mtlynch/tinypilot/issues/688

My setting.yml dont have any such setting:
```
tinypilot_repo_branch: 2.3.1
ustreamer_port: 8001
ustreamer_persistent: true
ustreamer_encoder: omx
ustreamer_format: uyvy
ustreamer_workers: 3
ustreamer_use_dv_timings: true
ustreamer_drop_same_frames: 30
```

According to the documentation

> Capture at 24 frames per second at 1080p resolution with less than 200 ms of latency.
> https://tinypilotkvm.com/

1080p is 1920×1080, so the resolution should not be the problem here...

taylor-schneider, 2022-03-14:
This issue is saying that resolution might be the problem and EDID is related: https://forum.tinypilotkvm.com/-102/no-signal-but-keyboard-and-mouse-are-working

Reviewing... this is way over my head

taylor-schneider, 2022-03-15:
At this point I am stuck... no idea... anyone else?

diego-tinypilot, 2022-03-15:
Hello @taylor-schneider - I'm sorry you are experiencing these issues with your TinyPilot.

* Have you tried changing the resolution of the target system? (e.g. set it to 800x600)
* Is it possible for you to try a different VGA to HDMI adapter?

diego-tinypilot, 2022-03-17:
Hi @taylor-schneider,

It looks like you have gone through a lot of troubleshooting yourself already, so it feels a bit embarrassing to ask you to do more testing, but if you could go through what I give you here, that would be awesome, and hopefully we can sort this out. It is recommended you reboot or power-cycle the TinyPilot in between each test. If you make any progress, and if it is not too much asking, please upload new logs.

From the logs, I can see what you have already pointed out, which is ustreamer is unable to start capturing.

`Mar 14 17:33:22 tinypilot ustreamer[698]: -- ERROR [2169484.567    stream] -- Unable to start capturing: Invalid argument
`

There are several scenarios for your problem:

* The VGA to HDMI adaptors you have tried may be incompatible with TinyPilot (remember the adaptors need to be USB powered, there are a few others that don't provide this and they are likely to fail with TinyPilot). Examples of compatible adaptors are the [FOINNEX](https://smile.amazon.com/dp/B07121Y1Z3/) and [uoeos](https://smile.amazon.com/gp/product/B088699FJV/).

* The HDMI cable you are using may be incompatible with TinyPilot. Although this would be a bit unlikely, because you have tried with a different PC using HDMI and it worked, I would still try a different high-quality HDMI cable (combined with one of the USB powered VGA to HDMI adapters I mentioned above).

* If at all possible, it would be also nice to test over DisplayPort instead of over VGA. I looked at your PC specifications, and I could see it also has a DisplayPort. If you have a DisplayPort to HDMI adaptor, please try to test it, it may just work.

* Does any video appears post-BIOS? I understand you do not have any option in the BIOS of the HP EliteDesk G2 Mini to change frame rate and/or resolution, but it would be wise to change resolution post BIOS (i.e. in the OS). If you lower the resolution in the OS, are you able to see anything at all?

* This is one of those cases where it is worth trying a different EDID. From your comment about it, it looks like you investigated this, but didn't try it. Here are the commands to do it. You would have to [SSH into the TinyPilot](https://tinypilotkvm.com/faq/ssh-credentials), and then run the commands below (you can just copy/paste them). Try to connect the TinyPilot again after this.

```

EDID="/home/ustreamer/edids/tc358743-edid.hex"

# Back up original EDID
cp --no-clobber "${EDID}" ~/tc358743-edid.hex.bak

# Set a new EDID
echo -ne "" | sudo tee "${EDID}" && \
  echo '00ffffffffffff0052628888008888882d1e0103' | sudo tee -a "${EDID}" && \
  echo '800000780aee91a3544c99260f50542540000100' | sudo tee -a "${EDID}" && \
  echo '0100010001000100010001010101d51b00505000' | sudo tee -a "${EDID}" && \
  echo '19400820b80080001000001eec2c80a070381a40' | sudo tee -a "${EDID}" && \
  echo '3020350040442100001e000000fc00546f736869' | sudo tee -a "${EDID}" && \
  echo '62612d4832430a20000000fd00323d0f2e0f0000' | sudo tee -a "${EDID}" && \
  echo '000000000000012a02030400de0d20a030581220' | sudo tee -a "${EDID}" && \
  echo '30203400f0b400000018e01500a0400016303020' | sudo tee -a "${EDID}" && \
  echo '3400000000000018b41400a050d0112030203500' | sudo tee -a "${EDID}" && \
  echo '80d810000018ab22a0a050841a3030203600b00e' | sudo tee -a "${EDID}" && \
  echo '1100001800000000000000000000000000000000' | sudo tee -a "${EDID}" && \
  echo '0000000000000000000000000000000000000000' | sudo tee -a "${EDID}" && \
  echo '00000000000000000000000000000045' | sudo tee -a "${EDID}" && \
  sudo v4l2-ctl --device=/dev/video0 --set-edid=file="${EDID}" --fix-edid-checksums

```

taylor-schneider, 2022-03-19:
@diego-tinypilot no worries, I will give these instructions a try. While I do, had some questions in response to the points you made:

> The VGA to HDMI adaptors you have tried may be incompatible with TinyPilot (remember the adaptors need to be USB powered, there are a few others that don't provide this and they are likely to fail [...]

- I don't understand the implication here. How would an adaptor drawing power from the usb port cause the tiny pilot (on the other end of an hdmi cable) to not receive the signal correctly? One of the adaptors I tried has a a vga port and a use port (for power) the other does not (it only has gva to usb).

> The HDMI cable you are using may be incompatible with TinyPilot. Although this would be a bit unlikely, because you have tried with a different PC using HDMI and it worked, I would still try a [...]

- The cable works when connected between other devices and tinypilot

> If at all possible, it would be also nice to test over DisplayPort instead of over VGA. I looked at your PC specifications, and I could see it also has a DisplayPort. If you have a DisplayPort to HDMI adaptor, please try to test it, it may just work.

- Unfortunately I dont have a DisplayPort to HDMI adaptor

> Does any video appears post-BIOS? 

- For a brief moment there is a scrambled signal. I believe it is the BIOS POST screen.

diego-tinypilot, 2022-03-22:
> I don't understand the implication here. How would an adaptor drawing power from the usb port cause the tiny pilot (on the other end of an hdmi cable) to not receive the signal correctly? 

The VGA spec originally didn’t require a power pin, but HDMI has required power from the start, so to convert from VGA to HDMI, you need power from outside the VGA adaptor (see screenshot below for reference, the power is coming from the USB port). There are some VGA adaptors that assume they’ll get power from the VGA port anyway, but those tend to cut corners with the VGA and HDMI spec and will have worse compatibility than powered adaptors.

[image]

> One of the adaptors I tried has a a vga port and a use port (for power) the other does not (it only has gva to usb).

Did you mean one of the adaptors you tried is similar to the picture above, but the other one is not USB powered? I'm confused by the last part of your sentence -> "it only has gva to usb" - did you mean to say it only has VGA to HDMI, without the USB part?

> For a brief moment there is a scrambled signal. I believe it is the BIOS POST screen.

Do you ever get to see anything else afterwards, like the OS booting, or anything else at all? Or is it just that brief moment of scrambled signal screen and then it goes to NO SIGNAL forever?

taylor-schneider, 2022-03-28:
@diego-tinypilot 

> Did you mean one of the adaptors you tried is similar to the picture above, but the other one is not USB powered? I'm confused by the last part of your sentence -> "it only has gva to usb" - did you mean to say it only has VGA to HDMI, without the USB part?

Apologies for the typos. Yes, One of the adaptors has an additional usb cable which I assume is drawing power while the other one did not.

> Do you ever get to see anything else afterwards, like the OS booting, or anything else at all? Or is it just that brief moment of scrambled signal screen and then it goes to NO SIGNAL forever?

I believe I see an image of the post (for an instant), then a scrambled mess (for a second or two), and then no signal forever. I am currently using the node but I will try to put it into maintenance mode to make sure this is true always.

P.S. I am still planning to do the EDID change. I am just taking some time to read up on the change. I am currently working remotely (for another two weeks) and want to understand how to revert the change remotely should anything go wrong.

diego-tinypilot, 2022-03-28:
If you followed the procedure I gave you above, and if the new EDID causes problems, you can restore the original EDID with these commands:

```bash
# Do this only if the steps above cause issues.
EDID="/home/ustreamer/edids/tc358743-edid.hex"
sudo cp ~/tc358743-edid.hex.bak "${EDID}"
sudo chown ustreamer:ustreamer "${EDID}"
```

mtlynch, 2022-03-28:
@taylor-schneider - It could be a defective or incompatible VGA adaptor. I sent you an email on March 22nd to see about getting you a replacement. Did you receive it?

Steroid-Factory, 2022-07-01:
I had an issue at work with this exact issue. I assembled an enclosure with 20 Pies that factory reset servers. I origionally ordered 14 FOINNEX vga to hdmi. A month later after protopying phase, I needed to get more, however the same FOINNEX in the new batch didn't display video correctly. I would only see Post, pxe menu, and loaded pxe image.  I tried 7 different vga to hdmi since I needed more more video cables. The issue is not with the vga to hdmi, but the capture card. I had good luck with Assletes Video Capture Cards and worked with all my vga to hdmi active adapters.

mtlynch, 2022-07-01:
Thanks, @Steroid-Factory!

When you saw capture failures, were you using an HDMI to CSI bridge or a USB capture dongle ?

Steroid-Factory, 2022-08-04:
@mtlynch, I was using a USB capture dongle. two important thing to note.
1. My answer is based on using TinyPilot software and my own Pi with Poe Hat. 
2. Most products on Amazon are manufactured in China and the sellers might have different suppliers. A USB Capture Card from the same Amazon link might be a different product a couple of months apart. The difference in suppliers/manufacturers will most likely differ in hardware circuitry of the dongle. This change may cause the VGA to HDMI cable not working. If video is not working on VGA to HDMI through a USB capture card, the issue most likely lies on the USB capture card end, rather than VGA to HDMI.

mtlynch, 2022-08-04:
All the low-cost HDMI to USB dongles I've seen are just the same [MS2109 chip](https://github.com/tiny-pilot/tinypilot/wiki/HDMI-Capture-Devices#macrosilicon-ms2109-capture-devices) on different PCBs, but it's possible some manufacturers are cutting corners and putting out buggy PCBs.

The Assletes brand seems to just be a brand name for the same generic capture dongle that others sell, so it's hard to say whether that brand actually controls the manufacturing process or if it was just luck, but I'd guess it's the latter.

closed 2022-08-04 by mtlynch