tiny-pilot/tinypilot #946: WebRTC frontend: provide fallback remote screen image

jotaen4tinypilot, 2022-03-30:
In contrast to the MJPEG stream, there is no default or fallback image, in case the video stream is not there (yet).

This might particularly be confusing on startup, because the screen appears all blank until the streaming connection is setup, which (on my device) takes up to 10 seconds.

jotaen4tinypilot, 2022-03-30:
Setting the [`poster` attribute](https://developer.mozilla.org/en-US/docs/Web/HTML/Element/video#attr-poster) to `poster="/stream?advance_headers=1"` on the video stream indeed works.

> A URL for an image to be shown while the video is downloading. If this attribute isn't specified, nothing is displayed until the first frame is available, then the first frame is shown as the poster frame.

In the video below you see how the resolution and colours improve at 0:05 (after the page refresh). This is where the WebRTC stream kicks in and replaces the MJPEG poster image. Then, at 0:13, the quality degrades again. This is where I shut down the Janus plugin on the server. So the poster image also works into “the other direction”, not just when initially loading.

https://user-images.githubusercontent.com/83721279/160890275-f3c6c6ae-8725-4417-8c1e-a05b8165576a.mov

Using the MJPEG as fallback is probably most resilient and user-friendly, because we fall back to something that the user can continue to work with.

On the other hand, users might have a reason why they want to use H264 (maybe bandwidth limitations?), so just falling back to MJPEG without telling them about it might not be what they expect. We could also make this configurable, though.

Anyway, I’m not sure we need to make a decision right away here, because I at least don’t have enough data points yet how reliable the Janus server plugin is. So maybe this is not that big of a problem in reality.

mtlynch, 2022-03-31:
Ha, that's neat behavior. I'm in favor of just silently falling back to MJPEG. I suspect that's what 98% of users want, and if we get feedback that users want more control over the fallback, we can add it later, but this seems like a win-win in terms of simplicity and reasonable behavior.

PR #950 (2022-04-21): Fallback to MJPEG when WebRTC is not enabled.
Depends on https://github.com/tiny-pilot/tinypilot/pull/965 being merged.

Resolves #946

---
This change is [[image]](https://reviewable.io/reviews/tiny-pilot/tinypilot/950)
closed 2022-04-21 by jdeanwallace