tiny-pilot/tinypilot #956: WebRTC frontend: Revisit control/command flow

jotaen4tinypilot, 2022-04-08:
After having set up the initial version of the WebRTC frontend based on Andre’s POC, we have learned (and still do so) more details about the control flow between frontend and server.

After we have finished writing the [guide](https://github.com/tiny-pilot/ustreamer/issues/1), we should backport our findings into [our code](https://github.com/tiny-pilot/tinypilot/blob/master/app/static/js/webrtc-video.js).

Just in case: this should wait until https://github.com/tiny-pilot/tinypilot/pull/950 is through, to avoid conflicts.

jotaen4tinypilot, 2022-04-21:
I realized that turning the `webrtc-video.js` file into a module (where all logic is encapsulated and exported as a class) would probably require us to *always* load [the respectice JS dependencies](https://github.com/tiny-pilot/tinypilot/blob/f3600c3468f699cfd3ca72c48b8aa58c2d3cbe3a/app/templates/index.html#L70-L80), even if WebRTC is turned off.

In JavaScript, top-level `import` statements can’t be conditional, so if we wanted to do `import { webrtcServer } from "app/js/webrtc-video.js"` in either `app.js` or `remote-screen.js`, it would lead to errors, since it wouldn’t find `Janus` to be defined anywhere.

Alternatively, we can use [dynamic, Promise-based imports](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Statements/import#dynamic_imports) like `import('app/js/webrtc-video.js').then(/*...*/)`. Those also work conditionally, but I find them rather obscure.

I’m basically wondering whether it’s worth doing all this. I feel that our bootstrapping procedure for WebRTC is a lot cleaner now, compared to the time when I created this ticket. The [`<remote-screen>` component](https://github.com/tiny-pilot/tinypilot/blob/master/app/templates/custom-elements/remote-screen.html) is completely unaware of the media sources, and has intuitive, well-defined APIs now. The `webrtc-video.js` script is drop-in, and doesn’t depend on any other specific page setup apart from the `<remote-screen>` instance. I actually think that’s not *too* bad of a structure.

What we still should do, however, is to incorporate the learnings, see initial comment. That in itself requires only a few minor modifications, and the additions of a few comments.

mtlynch, 2022-04-21:
I'm not in a rush to keep refactoring the WebRTC logic. If we end up doing more work on that code and find that the organization is creating problems, then we should create a new bug to figure out what to do about it, but I think what we have is pretty clean, given our needs.

I'm on board for integrating the lessons from [writing the uStreamer documentation](https://github.com/tiny-pilot/ustreamer/pull/2) now while it's still fresh in our heads.

closed 2022-04-22 by jotaen4tinypilot