tiny-pilot/tinypilot #958: Video disappears

c17r, 2022-04-11:
## Description

When I browser loads the TinyPilot UI, it will show the screen for a moment and then it disappears. If I'm quick enough to get the mouse over the video, it will stick around for a few seconds more, and I can see it showing the server mouse moving accordingly, but eventually the video will disappear again. I've updated the TinyPilot software but still getting the same behavior.

## What's the behavior that you expect?

That I can see the video so I can use the server.

## What's happening instead?

see above

## What are the steps to reproduce this behavior?

Just opening TinyPilot UI

## Screenshots

[image]

https://user-images.githubusercontent.com/148640/162817438-cdbefc8b-d4fe-4fc6-a03a-1cf316b233f3.mov

## Logs

http://logs.tinypilotkvm.com/0YWLZL5N

mtlynch, 2022-04-11:
Thanks for reporting this! We're refactoring the video code in preparation of [H264 support](https://github.com/tiny-pilot/tinypilot/issues/875), and this might be a regression.

Do you see any output in the [browser console](https://github.com/tiny-pilot/tinypilot/wiki/Troubleshooting-and-Diagnostics#browser-console-logs)?

Does the same thing happen if you open TinyPilot in an incognito window?

c17r, 2022-04-11:
No output in the browser console. Same behavior in an incognito window.

Is there an easy way to roll back to earlier versions? Happy to try that to debug this a bit.

mtlynch, 2022-04-11:
Sure, you can go to the TinyPilot repo and try going back a commit:

```bash
cd /opt/tinypilot
sudo su tinypilot bash -c "git checkout HEAD~1"
sudo service tinypilot restart
```

You can also try going back to a particular tag or commit like this:

```bash
cd /opt/tinypilot
sudo su tinypilot bash -c "git checkout 1.7.0"
sudo service tinypilot restart
```

mtlynch, 2023-05-01:
Closing as there hasn't been activity on this for awhile, but feel free to reopen with additional details.

closed 2023-05-01 by mtlynch