tiny-pilot/tinypilot #975: Purple screen - no login capability.  Login page presented normally on physical display

compumatter, 2022-05-06:
When I load the Tinypilot web interface I get a blank page.  

No errors in chrome
[image]
 
## Logs

http://logs.tinypilotkvm.com/VIVBs7iR

Thanks for looking into this.  FYI, I also tried updating the Tinypilot software but that did not resolve it.

mtlynch, 2022-05-09:
@compumatter - Thanks for reporting this!

It's strange because it is getting a video signal. Usually, TinyPilot either fails to capture any image at all (shows "No Signal"), or it captures an accurate image.

* What is the OS / hardware of the target computer? It looks like an Ubuntu purple.
* If you try connecting the display to a physical monitor instead of the TinyPilot, what appears on the screen?

@diego-tinypilot - Can you assist Jay here?

diego-tinypilot, 2022-05-09:
Hello @compumatter! I'll be glad to assist you. Would you answer Michael's questions before I proceed?

mtlynch, 2022-06-22:
Closed due to lack of response.

mtlynch, 2022-06-23:
Ah, glad to hear you got up and running!