tiny-pilot/tinypilot #982: Support mouse input on Android devices

mtlynch, 2022-05-30:
This [USB-C splitter](https://www.amazon.com/Multiport-Tuwejia-Thumderbolt-Converter-Projection/dp/B07G44M4S3) gives TinyPilot limited compatibility with Android phones.

One user reported that he was able to get video capture and keyboard input, but mouse input was not fully functional.

On a Pixel 4a, I'm able to reproduce the issue, though I'm not able to get video capture, as the Pixel 4a doesn't seem to support video output through USB-C.

I tried connecting a physical USB keyboard and mouse to the Pixel 4a through a USB hub, and both mouse and keyboard functioned normally on the phone. It's possible that Android only works with relative-positioned mice (TinyPilot's is absolute positioned). Or it's possible that there's something in TinyPilot's [USB HID descriptor](https://github.com/tiny-pilot/ansible-role-tinypilot/blob/bd7f56a8b81744d03f376c96e6ce4623a8832248/files/init-usb-gadget#L103L149) that's causing the Android to ignore its input.

man4567890, 2022-07-22:
I'm really interested in get it to do video output from an Android phone, how would you connect the phone and splitter, I cant wrap my mind around this, are there any diagrams available please

mtlynch, 2022-07-25:
There's a good diagram [in the forum](https://forum.tinypilotkvm.com/-261#post-6) showing how one user was able to achieve video capture.

It depends on the model of your Android device. Only certain devices support video output through the USB-C port.

man4567890, 2022-07-26:
Thank you, this is neat, I will order the splitter, Thanks again Michael