tiny-pilot/tinypilot #1002: Update Overhaul: Compress tarballs

jotaen4tinypilot, 2022-06-17:


jotaen4tinypilot, 2022-06-17:
Turns out, 3M of the tarball was the [`readme-assets`](https://github.com/tiny-pilot/tinypilot/tree/master/readme-assets) folder. Compression is still an easy win, but properly excluding non-needed files is more important.

mtlynch, 2022-06-22:
@jotaen4tinypilot - I think we should do this in milestone 1 rather than milestone 3. Changing the filenames and the unpack command is going to be more complicated once TinyPilot Community is relying on the new update flow.

jotaen4tinypilot, 2022-06-23:
Good point, I’ve moved it up to the 1st milestone.

Now that we have excluded the `readme-assets` folder, the net win isn’t that impressive anymore (819KB uncompressed vs 652KB compressed for the community bundle), but it’s an easy win and simple to implement, so I think it’s still better we have it right away.

PR #1029 (2022-06-27): Compress TinyPilot bundle with gzip
It's not a huge space savings right now, but it costs basically nothing to compress, and it might have a more meaningful impact in the future if our bundle size increases. It's much easier to change formats now before anything is using the bundles, so we're changing to .tgz.

| Bundle | Size |
|------------|----------|
| [tinypilot-community-20220622T1634Z-725b1a1.tar](https://bundles.tinypilotkvm.com/community/tinypilot-community-20220622T1634Z-725b1a1.tar) | 800 KB |
| [tinypilot-community-20220624T1527Z-56e4671.tgz](https://bundles.tinypilotkvm.com/commun [... cut]
closed 2022-06-27 by mtlynch