tiny-pilot/tinypilot #995: Use f-strings in places that pylint recommends

mtlynch, 2022-06-17:
New versions of pylint flag places where we can use [Python f-strings](https://realpython.com/python-f-strings/) instead of the legacy string formatting options.

We currently have pylint's check for this disabled because of our legacy code that uses the old formatting style, but we should replace those instances with modern f-strings and delete [the suppression in .pylintrc](https://github.com/tiny-pilot/tinypilot/blob/d89da5e8513fbf15b3af8b231b0522ff1350fc25/.pylintrc#L26).

ezinneanne, 2022-06-17:
Hello, can I be assigned to this?

mtlynch, 2022-06-17:
Sure, that would be great!

This depends on #993, which is not yet merged into master, but you can fork from the `pylint-2.14.2` branch and make these fixes.

mtlynch, 2022-06-20:
@ezinneanne - #993 is now merged, so you don't have to do anything special with branch forking. You can open a regular PR against the `master` branch.

jmunoz94, 2022-10-12:
Hey, @mtlynch: I see that this issue is assigned, but it has been a while and it seems there is no progress on it (ex: ezinneanne has not even forked the repository). Can I get this assigned to me/work on this?

mtlynch, 2022-10-12:
Sure, @jmunoz94 - assigned.

PR #1143 (2022-10-14): Use f-strings in places pylint recommends
Fixes #995

# Background

tinypilot currently uses old c-style formatting. In order to avoid any complaints from pylint, the `consider-using-f-string` check was disabled. This PR attempts to fix that by updating potentially problematic lines to use f-strings and then removing this check from the list of disabled rules.

# Changes

* Change c-style formatting by f-strings.
* Remove `consider-using-f-string` from the disabled rules in `.pylintrc`.
[image]
closed 2022-10-14 by mtlynch