tiny-pilot/tinypilot #1015: quick-install fails while installing pkg-resources==0.0.0

scirelli, 2022-06-20:
## Description
Ran the quick install. It errored out with 
> Could not find a version that satisfies the requirement pkg-resources==0.0.0

## What's the behavior that you expect?

tinypilot to install and restart the rpi.

## What's happening instead?

quick-install errored out with
> Could not find a version that satisfies the requirement pkg-resources==0.0.0

## What are the steps to reproduce this behavior?
1. Ran the quick install step found in the README

## Screenshots
[image]

## Logs
```
...
+ tee --append /var/tmp/tmp.MFI5mvlHQc.yml
ustreamer_format: jpeg
+ add_setting_if_undefined ustreamer_resolution 1920x1080
+ local key=ustreamer_resolution
+ local value=1920x1080
+ grep --silent '^ustreamer_resolution:' /var/tmp/tmp.MFI5mvlHQc.yml
+ tee --append /var/tmp/tmp.MFI5mvlHQc.yml
+ echo 'ustreamer_resolution: 1920x1080'
ustreamer_resolution: 1920x1080
+ echo 'Final install settings:'
Final install settings:
+ cat /var/tmp/tmp.MFI5mvlHQc.yml
ustreamer_port: 8001
ustreamer_persistent: true
ustreamer_encoder: hw
ustreamer_format: jpeg
ustreamer_resolution: 1920x1080
+ HAS_PRO_INSTALLED=0
++ dirname bash
+ SCRIPT_DIR=.
+ '[' . = . ']'
+ SCRIPT_DIR=/opt/tinypilot
+ TINYPILOT_README=/opt/tinypilot/README.md
+ '[' -f /opt/tinypilot/README.md ']'
+ '[' 0 = 1 ']'
+ sudo apt-get update --allow-releaseinfo-change-suite
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists... Done
+ sudo apt-get install -y git libffi-dev libssl-dev python3-dev python3-venv
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.30.2-1).
libffi-dev is already the newest version (3.3-6).
libssl-dev is already the newest version (1.1.1n-0+deb11u2+rpt1).
The following additional packages will be installed:
  javascript-common libjs-jquery libjs-sphinxdoc libjs-underscore libpython3-dev libpython3.9-dev python-pip-whl python3-distutils python3-lib2to3 python3.9-dev python3.9-venv
Suggested packages:
  apache2 | lighttpd | httpd
The following NEW packages will be installed:
  javascript-common libjs-jquery libjs-sphinxdoc libjs-underscore libpython3-dev libpython3.9-dev python-pip-whl python3-dev python3-distutils python3-lib2to3 python3-venv python3.9-dev python3.9-venv
0 upgraded, 13 newly installed, 0 to remove and 0 not upgraded.
Need to get 6,380 kB of archives.
After this operation, 19.4 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian bullseye/main armhf python-pip-whl all 20.3.4-4+rpt1 [2,002 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf javascript-common all 11+nmu1 [6,260 B]
Get:3 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libjs-jquery all 3.5.1+dfsg+~3.5.5-7 [315 kB]
[... 121 lines cut ...]
  Downloading https://www.piwheels.org/simple/ansible/ansible-2.9.10-py3-none-any.whl (16.2 MB)
     |████████████████████████████████| 16.2 MB 22 kB/s
Collecting cffi==1.14.4
  Downloading cffi-1.14.4.tar.gz (471 kB)
     |████████████████████████████████| 471 kB 778 kB/s
  Preparing metadata (setup.py) ... done
Collecting cryptography==35.0.0
  Downloading cryptography-35.0.0.tar.gz (559 kB)
     |████████████████████████████████| 559 kB 1.2 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting Jinja2==2.11.2
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-2.11.2-py2.py3-none-any.whl (127 kB)
     |████████████████████████████████| 127 kB 365 kB/s
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/ma
rkupsafe/
Collecting MarkupSafe==1.1.1
  Downloading MarkupSafe-1.1.1.tar.gz (19 kB)
  Preparing metadata (setup.py) ... done
ERROR: Could not find a version that satisfies the requirement pkg-resources==0.0.0 (from versions: none)
ERROR: No matching distribution found for pkg-resources==0.0.0
WARNING: You are using pip version 21.3.1; however, version 22.1.2 is available.
You should consider upgrading via the '/opt/tinypilot-updater/venv/bin/python3 -m pip install --upgrade pip' command.
```

mtlynch, 2022-06-21:
@scirelli - Does this continue happening if you attempt to run `quick-install` again?

I'm wondering if the `pkg-resources` error is a red herring, and it's actually a side effect of your system getting disconnected from the mirror while attempting to install the MarkupSafe package:

```text
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/ma
rkupsafe/
```

scirelli, 2022-06-21:
> @scirelli - Does this continue happening if you attempt to run `quick-install` again?
> 
> I'm wondering if the `pkg-resources` error is a red herring, and it's actually a side effect of your system getting disconnected from the mirror while attempting to install the MarkupSafe package:
> 
> ```
> WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/ma
> rkupsafe/
> ```

Maybe, but I've tried multiple times with the same error. I was going to remove that dependency and see what happens. Will the project still function without it?

scirelli, 2022-06-22:
@mtlynch I removed `pkg-resources` and installed. Restarted and it seems like it's working. I'll connect it to something to be sure. This is a pi 0w v1 btw.

[image]

closed 2022-06-22 by mtlynch

scirelli, 2022-06-22:
That's not prob. I only care about sending KB data.

mtlynch, 2022-06-22:
Ah, before TinyPilot, I wrote a keyboard-only prototype called [Key Mime Pi](https://mtlynch.io/key-mime-pi/). It's prototype-quality code, but it's simpler than TinyPilot, so you might be interested in that as an alternative.

PR #1032 (2022-06-28): Remove pkg-resources from list of quick-install requirements
According to this thread, this line was always erroneous and we should just remove it: https://github.com/pypa/pip/issues/4022

It doesn't seem to harm anything on Raspberry Pi OS, but it causes problems on other distros.

Fixes #1031
Fixes #1015

---
This change is [[image]](https://reviewable.io/reviews/tiny-pilot/tinypilot/1032)