tiny-pilot/tinypilot #1026: Pasting Random Cancelation

scirelli, 2022-06-23:
## Description
### Environment 
Hardware device: r-pi 0w v1
Broswer: Chrome
<br/>
When pasting text of any size input on the receiving end randomly stops. I say random because I can't tell what causes it to end the stream. Sometimes I'll be able to paste, I'd say 100 characters, other times it stops after maybe 15-20. I'm not sure why it fails, and there's no feedback in the UI.

## What's the behavior that you expect?
Pasting any size text to be streamed to remote computer. 

## What's happening instead?
Characters stop getting typed after a random amount sent.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
3. Click on menu item to paste text
4. Paste text
5. Not all text shows up on remote computer

## Screenshots
Nothing to show.

## Logs
I will have to reproduce this again and copy out the logs. It consistently happens so I'll be able to reproduce it again tonight.

mtlynch, 2022-06-24:
It's hard to say what's going on without the logs. Can you see if you can capture logs after it happens?

In addition the the device logs, it would be good to see the [browser console logs](https://github.com/tiny-pilot/tinypilot/wiki/Troubleshooting-and-Diagnostics#browser-console-logs) as well in case it's failing before the browser requests ever reach the TinyPilot device.

scirelli, 2022-06-24:
Attached

scirelli, 2022-06-24:
I think it might have to do with the network. Happened less when the sending device was wired

mtlynch, 2022-06-27:
I'm not seeing any hints in the logs. There were these lines:

```text
Jun 23 09:50:55 rpi0w-keyboard python[1048]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
```

But those occurred hours before you collected the logs, so I'm assuming that's not part of you attempt to reproduce this.

I tried pasting a long sequence of text from *Moby Dick*, and I was able to reproduce the issue on both a real device and a dev system. The paste seems to fail after around 2000 characters in dev mode.

I added debug logging to the JS, and I see it emit all the characters from the browser, but the backend doesn't receive them. At a certain point, the backend just stops processing keystroke messages, and there's no error.

My best guess is that we're causing unexpected behavior by flooding the websockets channel, though I can't find anything in [socket.io's docs](https://socket.io/docs/v3/emitting-events/) about that. I tried upgrading to the socket.io JS client v4.5.0, and I see the same behavior.

I suspect that we could solve this by buffering the keystrokes instead of sending them immediately, so instead of [sending all keystrokes as we see them](https://github.com/tiny-pilot/tinypilot/blob/f62906aa1d1dea187c060cfa5af1437a4380bc8b/app/static/js/keystrokes.js), we could send a maximum of N keystrokes until we received acknowledgement from the backend that the keystroke was processed successfully.

scirelli, 2022-07-08:
I think it might have to do with how the fast the receiving computer reads keys strokes. To fast and it does't get the release stroke, or possible bytes get miss aligned? Not sure, but slowing key strokes down seemed to work.

db39, 2023-06-13:
A [user on the forum](https://forum.tinypilotkvm.com/-681#post-2) has also encountered this issue when pasting larger quantities of text. Their current workaround of batching their text in multiple paste actions works - but is obviously frustrating.

scirelli, 2023-06-13:
Just to add some info. I was using this on the Nintendo Switch, the Switch has some arbitrary keystroke entry delay. You can not type fast on the Switch.

mtlynch, 2023-07-06:
This issue could be just a bug in our socket.io client or server libraries. As a first step, we should upgrade to the latest and greatest versions of both ends and see if that resolves the issue.

mtlynch, 2023-07-07:
@jotaen4tinypilot - Oh, yeah. That's a good idea.

When I implemented paste, I debated between doing the text-to-keystroke conversions server-side or client-side, and I did them client-side just for the sake of expediency. But server-side, we could have better testing, and it makes more sense to send everything in one big request than to flood the websockets channel with a million little keystroke requests.

mtlynch, 2023-07-07:
Cool, that looks better than what we're currently doing for paste!

If we move server-side, I think we should probably just skip the character echo.

The main purpose of the character echo is to give fast visual feedback for keystrokes to compensate for the jarring feeling of typing with a delay. It's not as important for pasting text.

The fact that we currently echo pasted text in the keystroke history is just a side-effect of the paste functionality piggybacking on existing plumbing, not something we chose to do deliberately.

jdeanwallace, 2023-07-25:
Quick update

> This issue could be just a bug in our socket.io client or server libraries. As a first step, we should upgrade to the latest and greatest versions of both ends and see if that resolves the issue.

Updating our socket.io libraries on both the client and server side, doesn't seem to fix the issue
* Server side:
    ```diff
    # requirements.txt
    -Flask-SocketIO==5.0.1
    +Flask-SocketIO==5.3.4
    -python-engineio==4.0.1
    -python-socketio==5.1.0
    +python-engineio==4.5.1
    +python-socketio==5.8.0
    ```
* Client side:
    ```diff
    # app/templates/index.html
    -    <script src="/third-party/socket.io/3.1.3/socket.io.min.js"></script>
    +    <script src="/third-party/socket.io/4.7.1/socket.io.min.js"></script>
    ```

---

> I tried pasting a long sequence of text from Moby Dick, and I was able to reproduce the issue on both a real device and a dev system. The paste seems to fail after around 2000 characters in dev mode.

I enabled socketio logging on the server side and retried the above test:

```diff
# app/socket_api.py
-socketio = flask_socketio.SocketIO()
+socketio = flask_socketio.SocketIO(engineio_logger=True, logger=True)
```

From the logs, you can see the moment the paste starts failing:

```
2023-07-25 14:40:57.478 engineio.server INFO kLYh-qEY47rUf5lOAAAA: Received packet MESSAGE data 2648["keystroke",{"metaLeft":false,"metaRight":false,"altLeft":false,"altRight":false,"shiftLeft":false,"shiftRight":false,"ctrlLeft":false,"ctrlRight":false,"key":" ","code":"Space"}]
received event "keystroke" from jZE8Oj_LMPZS5xZqAAAB [/]
2023-07-25 14:40:57.478 socketio.server INFO received event "keystroke" from jZE8Oj_LMPZS5xZqAAAB [/]
2023-07-25 14:40:57.491 socket_api      DEBUG [SENSITIVE] received keystroke message: {'metaLeft': False, 'metaRight': False, 'altLeft': False, 'altRight': False, 'shiftLeft': False, 'shiftRight': False, 'ctrlLeft': False, 'ctrlRight': False, 'key': ' ', 'code': 'Space'} [/SENSITIVE]
2023-07-25 14:40:57.493 hid.write       DEBUG [SENSITIVE] writing to HID interface /dev/null: 0x00 0x00 0x2c 0x00 0x00 0x00 0x00 0x00 [/SENSITIVE]
2023-07-25 14:40:57.512 hid.write       DEBUG [SENSITIVE] writing to HID interface /dev/null: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 [/SENSITIVE]
kLYh-qEY47rUf5lOAAAA: Client is gone, closing socket
2023-07-25 14:40:57.544 engineio.server INFO kLYh-qEY47rUf5lOAAAA: Client is gone, closing socket
2023-07-25 14:40:57.548 socket_api      INFO Client jZE8Oj_LMPZS5xZqAAAB disconnected
kLYh-qEY47rUf5lOAAAA: Client is gone, closing socket
2023-07-25 14:40:57.548 engineio.server INFO kLYh-qEY47rUf5lOAAAA: Client is gone, closing socket

<---------------------- PASTE STARTS FAILING NOW ---------------------->

kLYh-qEY47rUf5lOAAAA: Received packet MESSAGE data 2649["keystroke",{"metaLeft":false,"metaRight":false,"altLeft":false,"altRight":false,"shiftLeft":false,"shiftRight":false,"ctrlLeft":false,"ctrlRight":false,"key":"t","code":"KeyT"}]
2023-07-25 14:40:57.549 engineio.server INFO kLYh-qEY47rUf5lOAAAA: Received packet MESSAGE data 2649["keystroke",{"metaLeft":false,"metaRight":false,"altLeft":false,"altRight":false,"shiftLeft":false,"shiftRight":false,"ctrlLeft":false,"ctrlRight":false,"key":"t","code":"KeyT"}]
received event "keystroke" from None [/]
2023-07-25 14:40:57.549 socketio.server INFO received event "keystroke" from None [/]
None is not connected to namespace /
2023-07-25 14:40:57.549 socketio.server WARNING None is not connected to namespace /
kLYh-qEY47rUf5lOAAAA: Received packet MESSAGE data 2650["keystroke",{"metaLeft":false,"metaRight":false,"altLeft":false,"altRight":false,"shiftLeft":false,"shiftRight":false,"ctrlLeft":false,"ctrlRight":false,"key":"o","code":"KeyO"}]
2023-07-25 14:40:57.550 engineio.server INFO kLYh-qEY47rUf5lOAAAA: Received packet MESSAGE data 2650["keystroke",{"metaLeft":false,"metaRight":false,"altLeft":false,"altRight":false,"shiftLeft":false,"shiftRight":false,"ctrlLeft":false,"ctrlRight":false,"key":"o","code":"KeyO"}]
received event "keystroke" from None [/]
2023-07-25 14:40:57.750 socketio.server INFO received event "keystroke" from None [/]
None is not connected to namespace /
```

From what I understand, the "Client is gone, closing socket" message means that the client-server connection couldn't be maintained the the connection was dropped. I suspect that the many `keystroke` messages are blocking the server from responding to the connection pings.

mtlynch, 2023-07-25:
Ah, good finding.

Yeah, it seems like we're probably just sending too fast and flooding the channel, though that result still surprises me a little. It doesn't seem like we're sending so quickly that we'd essentially DoS the channel, but I also don't think it's worth digging into deeper.

I think we should proceed with Jan's proposal of sending the characters via HTTP and converting them to keystrokes server-side.

Can you check in the socket.io version change PR, though? Might as well be on latest and greatest.

jdeanwallace, 2023-07-26:
@mtlynch 
> I think we should proceed with Jan's proposal of sending the characters via HTTP and converting them to keystrokes server-side.

I'm not totally convinced that we are reaching some (possibly undocumented) buffer size limit because the default buffer size limit (in polling mode) seems to be 1MB, which is [>1million characters](https://stackoverflow.com/a/48698806/3769045).

Similar to @jotaen4tinypilot's idea, instead of sending the pasted text via a HTTP API I propose adding a bulk verison of the [`keystroke` endpoint](https://github.com/tiny-pilot/tinypilot/blob/master/app/socket_api.py#L20) that accepts multiple keystrokes at once instead of just one at a time.

mtlynch, 2023-07-26:
Yep, I agree with a bulk endpoint to send the full paste in a single request.

I think we should make it a regular HTTP POST rather than a websockets endpoint because HTTP POST is easier to debug and reason about, and we don't need the performance benefit of websockets.

jdeanwallace, 2023-08-11:
I'm having trouble resolving this issue for when the client sends many keystrokes right after each other (which just happens to be exactly how our paste functionality works):
https://github.com/tiny-pilot/tinypilot/blob/330978abc7288b64d61f279fd687816f3cfde3f6/app/static/js/app.js#L258-L260

I've tried a couple things, which didn't work:
1. Send keystrokes in series, waiting for a callback after each keystroke
    This made the pasting functionality extremely slow, so not a viable option.
1. Calling `socketio.sleep()` in a loop while waiting for a daemon process to finish writing to the HID interface
    The same bug occurred (i.e. "Client is gone, closing socket")
1. Putting the `keystroke` and `keyRelease` socketio endpoint into their own namespace (in case they were blocking the normal ping/pong keepalive messages)
    The same bug occurred
1. Force the ping/pong messages to fire every second to keep the client/server connection alive
    I could see the ping/pong messages being logged on the server, between all the keystroke messages, but the client/server connection still dies after a while (i.e., same bug occurred)
1. Don't even write to the HID interface and just call `socketio.sleep(1)` instead
    Sometimes I could get through all the pasted keystrokes without the client/server disconnecting, but repeated tests resulted in the same bug still occuring 🤯 
    I don't get it, calling socketio.sleep() should be unblocking our single threaded server.

mtlynch, 2023-08-11:
@jdeanwallace - Nothing comes to mind, but can you summarize what we know about the bug so far?

My original hypothesis was that we were flooding the websockets channel, but it seems like that's unlikely given further testing. 

Now it seems like something is causing the server to stop responding to websockets / HTTP requests while it's processing a long sequence of keystrokes. Do we know what's causing it to stop responding?

>Don't even write to the HID interface and just call socketio.sleep(1) instead

Where did we skip the write? Within the spawned process in `hid/write.py`? Or before we spawn a new process?

My suspicion is that we're doing something wrong in [`app/hid/write.py`](https://github.com/tiny-pilot/tinypilot/blob/330978abc7288b64d61f279fd687816f3cfde3f6/app/hid/write.py). We're [spawning a new process for each HID event](https://github.com/tiny-pilot/tinypilot/blob/330978abc7288b64d61f279fd687816f3cfde3f6/app/hid/write.py#L85), so spawning so many short-lived processes seems like it might interfere with Flask staying responsive.

In particular, this snippet of the docs makes me think we don't fully understand how Flask handles multiple processes:

>Flask, as a WSGI application, uses one worker to handle one request/response cycle. When a request comes in to an async view, Flask will start an event loop in a thread, run the view function there, then return the result.

>Each request still ties up one worker, even for async views. The upside is that you can run async code within a view, for example to make multiple concurrent database queries, HTTP requests to an external API, etc. However, the number of requests your application can handle at one time will remain the same.

https://flask.palletsprojects.com/en/2.3.x/async-await/#performance

I'm wondering if we need to be using `eventlet`-specific threading APIs the same way we do to [avoid locking up the server in `update_logs.py`](https://github.com/tiny-pilot/tinypilot/blob/330978abc7288b64d61f279fd687816f3cfde3f6/app/update_logs.py#L84). Like maybe we should be using [`tpool`](http://eventlet.net/doc/threading.html#tpool-simple-thread-pool)?

But before we get into trying new fixes, I'd like to just summarize where we are now. Partly because it's a good rubber duck debugging exercise, and partly so @jotaen4tinypilot can weigh in when he gets back.

mtlynch, 2023-08-14:
@jotaen4tinypilot - Any ideas?

mtlynch, 2023-08-15:
>It would be worth to dive deeper into the threading setup of Flask and Flask-SocketIO, just to gain a better understanding of why the socket endpoint freezes at some point in situations where the server is busy processing things. For example, why does Flask-SocketIO becomes “blocked” internally, while the rest of the Flask server stays responsive (e.g., the HTTP endpoints)? Or, why is it that socketio.sleep doesn’t have any effect? To me, it looks like there is something off with our threading setup, or with the way Flask and Flask-SocketIO are wired in regards to threading.

This seems like the best route to me.

@jdeanwallace - Can you create a minimal repro with Python-SocketIO of this issue?

I'm thinking we'd create an app like the following:

1. App spins up a python-socketio server like TinyPilot's
2. App serves a single `index.html` page
3. `index.html` page has JS that creates a websockets connection and updates the page if the websockets connection ever breaks
4. `index.html`'s JS sends websockets messages to the backend that cause some blocking process (e.g., sleep, write a 500 MB file)

Hopefully, that reproduces the error in a way that's pretty portable and simple (i.e., doesn't have to be tested on a real Pi).

I suspect that once we create a minimal repro, the solution will soon follow. If it doesn't, the author of python-socketio [does short-term consulting](https://blog.miguelgrinberg.com/post/hire-me), so I think we could show him the repro, and he would be able to identify the issue pretty quickly.

jdeanwallace, 2023-08-15:
@mtlynch 

> Can you create a minimal repro with Python-SocketIO of this issue?

Sure, but why not [Flask-SocketIO](https://github.com/miguelgrinberg/Flask-SocketIO)? They're both maintained by same author and we're using Flask-SocketIO (which admittedly does use Python-SocketIO under the hood).

mtlynch, 2023-08-15:
@jdeanwallace - Right, sorry, that's what I meant. I forgot that there was a layer on top of Python-SocketIO.

jdeanwallace, 2023-08-15:
@mtlynch - You can find the minimal repro project here:
https://github.com/tiny-pilot/tinysocket

I'll leave it to you to assign permissions and licenses.

mtlynch, 2023-08-15:
@jdeanwallace - Nice! I like the name. Good clean implementation, too.

I'd like to adjust it a little bit to get confidence the issue it reproduces is the same as what we're observing here, so I filed https://github.com/tiny-pilot/tinysocket/issues/2.

jdeanwallace, 2023-08-23:
# 2023-08-23 – What we know so far

We created a minimal repro of the reported issue via https://github.com/tiny-pilot/tinysocket, which removed many TinyPilot complexities from the issue and helped us better understand Flask-SocketIO behavior. However, the current repro might just be the upper limit of what Flask-SocketIO can handle. The repro does display similar symptoms as the reported issue (i.e., the client being disconnected then reconnected when sending many WebSocket messages), but under different conditions (i.e., requires 30x more WebSocket messages).

## Current understanding:
* The TinyPilot Flask server is single threaded
* Both HTTP requests and WebSocket messages are handled by the single threaded Flask server
* On every `keystroke` message that the WebSocket server receives, the server spawns a process that blocks the main thread for up to 500ms before responding.
    https://github.com/tiny-pilot/tinypilot/blob/edb44415699c1a35925933026e9d86a26f032104/app/hid/write.py#L81-L86
* Looking at the [Flask-SocketIO example app](https://github.com/miguelgrinberg/Flask-SocketIO/blob/ea33de9baf62281d88919d8fc6b397dbc439e4cf/example/app.py#L109), it seems like TinyPilot isn't making proper use of `eventlet`'s coroutines (i.e., green threads) because we never actually call [`socketio.start_background_task`](https://flask-socketio.readthedocs.io/en/latest/api.html#flask_socketio.SocketIO.start_background_task). This suggests that we're blocking the main Flask thread.

## Next steps:
1. As suggested by https://github.com/tiny-pilot/tinysocket/issues/2, we mimic the TinyPilot behavior more closely by spawning processes that block for up to 500ms to repro the issue with less WebSocket messages in order to match the conditions of the reported issue.
2. Fix/improve the confirmed repro in step (1) by properly utilizing `eventlet`'s green threads

mtlynch, 2023-08-23:
Thanks, @jdeanwallace! This is a helpful writeup!

jdeanwallace, 2023-08-28:
# 2023-08-28 – What we know so far

We recreated the original issue using a [minimal Flast-SocketIO setup](https://github.com/tiny-pilot/tinysocket/blob/39007a5d70e8062110f2614acd28bd3ff01a3a1d/app.py#L22-L24) which spawns a processes that blocks the main server thread for up to 500ms on each WebSocket message.

With a simplified repro in-hand we tried to solve the original issue by making proper use of `eventlet`'s green/virtual threads. Unfortunately, we saw the same behavior even when using `eventlet` in the way that Flask-SocketIO expects us to. So at least we know that using processes instead of threads didn't cause this issue.

## Current understanding
* Flask-SocketIO doesn't give priority to internal PING/PONG messages. So if the client floods the server with a bunch of custom WebSocket messages that it can't process within the ping timeout, the server perceives the connection as broken and disconnects the client.
* By rate-limiting the number of custom WebSocket messages sent by the frontend, we're able to ensure that the server is able to process the internal PONG messages in between a potentially long list of custom WebSocket messages. A proof-of-concept was created here:
    *  https://github.com/tiny-pilot/tinysocket/tree/throttle-js

## Next steps
1. Implement rate-limiting of "keystroke" events on the TinyPilot frontend, that sends a maximum of ~20 keystrokes per second to the backend
    * In the future, we'd ideally want to use a dynamic rate limit based on how fast the server is able to process WebSocket messages

closed 2023-09-19 by jdeanwallace

mtlynch, 2023-09-27:
I tried a TinyPilot Pro build with this fix included. As a test, I tried generating five paragraphs (4300 characters) [of lorem ipsum text](https://loremipsum.io/generator/?n=5&t=p) and pasted it into [an MD5 calculator](https://xorbin.com/tools/md5-hash-calculator) on the client machine and the target machine.

[image]

[image]

Both versions had the same md5, so it's looking good!