tiny-pilot/tinypilot #1027: Update Overhaul: final QA and testing

jotaen4tinypilot, 2022-06-23:
After the last item of [the second milestone](https://github.com/tiny-pilot/tinypilot-pro/issues/445) is complete, we want to do a thorough testing phase before merging the `update-overhaul` branch into `master`.

We still need to specify the testing procedure and scenarios in more detail, but on a high level, we should do:

- Test update path from an existing TinyPilot installation
- Test new installation on fresh device
- Identify points of failure and verify that it behaves correctly
- Make sure that we don’t break existing systems, e.g. by accidentally overriding settings or configs

Only after all testing is done, we merge the `update-overhaul` branch into `master`. We should probably also schedule this wisely, to make sure that we have sufficient dev and support coverage in the following days of the release.

jotaen4tinypilot, 2022-07-22:
It looks like we are about to wrap up [Milestone 2](https://github.com/tiny-pilot/tinypilot-pro/issues/445) and make things happen 🎉 

I’ve revisited the initial bullet points of this ticket, and tried to make them more concrete. @mtlynch @jdeanwallace Please feel free to add or adjust the items above, if you can think of any other scenario that’s not covered yet.

Thanks also @jdeanwallace for referencing this ticket from the PRs: that was pretty helpful for keeping an overview.

I’m wondering how we go about this ticket: it could make sense if two (or even three) people independently do the entire QA, just so that we have some extra safety?

As for the “tbd” regarding the Community bundles (see above): the most minimal-invasive way I can think of would be to upload a `.tgz` bundle built from the `update-overhaul` branch, and manually edit the `LATEST` file to point to it. That way, we don’t have to touch the update mechanism itself, and everything would exactly behave as it does for real.

mtlynch, 2022-07-22:
> As for the “tbd” regarding the Community bundles (see above): the most minimal-invasive way I can think of would be to upload a .tgz bundle built from the update-overhaul branch, and manually edit [...]

Yeah, that sounds good to me.

mtlynch, 2022-07-29:
@jdeanwallace - Is the task list ready for me to review? Just want to make sure we've merged everything over from #1046

jdeanwallace, 2022-07-29:
@mtlynch - Yes, you can review the task list. I'm just still finalizing https://github.com/tiny-pilot/tinypilot/pull/1046 before the final round of testing.

mtlynch, 2022-07-29:
@jdeanwallace - Okay, I've split it into individual tasks. What's gating the start of testing?

jdeanwallace, 2022-08-01:
@mtlynch - I was hoping to merge https://github.com/tiny-pilot/tinypilot/pull/1076 & spot check the current testing instructions before getting others involved in testing.

mtlynch, 2022-08-01:
@jdeanwallace - Okay, sounds good.

jdeanwallace, 2022-08-01:
@mtlynch - I'm happy with the state of https://github.com/tiny-pilot/tinypilot/pull/1046 to let others start testing.

mtlynch, 2022-08-02:
@jdeanwallace - Let's hold off on starting the tests until we fix #1081 and #1082.

jotaen4tinypilot, 2022-08-12:
Just double-checking, since I was off: is it safe to start testing, or are there still open, ongoing issues?

jdeanwallace, 2022-08-12:
> Just double-checking, since I was off: is it safe to start testing, or are there still open, ongoing issues?

Hey @jotaen4tinypilot, looking at the [project milestone view](https://github.com/orgs/tiny-pilot/projects/6/views/2), we're still waiting on [one issue](https://github.com/tiny-pilot/ansible-role-tinypilot-pro/issues/119) that krept in before we can start the final update-overhaul test/review.

mtlynch, 2022-08-15:
@jdeanwallace - Any blockers to begin testing at this point?

jotaen4tinypilot, 2022-08-16:
fyi, I have removed the point below, since I think we have this already covered separately (see first section)

> ### Hardware compatibility
> - [ ] Check that the installation works on both Voyager and Hobbyist device

jotaen4tinypilot, 2022-08-17:
It all seems to be looking good so far… 🎉 

- I don’t have a (working) Hobbyist device at hand, because with mine the video capture chip is defunct. I’m not sure it makes sense for me to take over the two Hobbyist test items, because I wouldn’t be able to verify whether the video setup and configs are working.
- Regarding the “test upgrade path to Pro from Community”, I realized that I actually don’t know how to do this, because I usually just flash an SD card with a disk image for setting up a Pro device. But that’s obviously not an option in this case. Is there an official way to do this, or would I just download and run Pro’s quick-install script from the private repo?
- One last-minute question occurred to me, [about where to host the `get-tinypilot.sh` script](https://github.com/tiny-pilot/ansible-role-tinypilot/issues/216).

jdeanwallace, 2022-08-18:
@jotaen4tinypilot 

> * I don’t have a (working) Hobbyist device at hand, because with mine the video capture chip is defunct. I’m not sure it makes sense for me to take over the two Hobbyist test items, because I wouldn’t be able to verify whether the video setup and configs are working.

Okay cool, I'll take on these two Hobbyist tests items.

> * Regarding the “test upgrade path to Pro from Community”, I realized that I actually don’t know how to do this, because I usually just flash an SD card with a disk image for setting up a Pro [...]

I've sent you an email regarding the Pro installation steps.

jotaen4tinypilot, 2022-08-19:
Pending https://github.com/tiny-pilot/tinypilot-pro/pull/541, the QA is all done.

As closing remark from my side, I’d feel confident enough to close this ticket, so I feel we have tested and considered everything quite thoroughly.