tiny-pilot/tinypilot #1031: Installation on Armbian/Ubuntu 22.04 fails

spitfire, 2022-06-26:
I was trying to install it onm Orange Pi One running Armbian 22.05 Jammy (based on Ubuntu 22.04) and it failed with the following:

```
six==1.15.0'
+ pip install -r requirements.txt
Collecting ansible==2.9.10
  Using cached ansible-2.9.10.tar.gz (14.2 MB)
  Preparing metadata (setup.py) ... done
Collecting cffi==1.14.4
  Using cached cffi-1.14.4.tar.gz (471 kB)
  Preparing metadata (setup.py) ... done
Collecting cryptography==35.0.0
  Using cached cryptography-35.0.0.tar.gz (559 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting Jinja2==2.11.2
  Using cached Jinja2-2.11.2-py2.py3-none-any.whl (125 kB)
Collecting MarkupSafe==1.1.1
  Using cached MarkupSafe-1.1.1.tar.gz (19 kB)
  Preparing metadata (setup.py) ... done
ERROR: Could not find a version that satisfies the requirement pkg-resources==0.0.0 (from versions: none)
ERROR: No matching distribution found for pkg-resources==0.0.0
WARNING: You are using pip version 21.3.1; however, version 22.1.2 is available.
You should consider upgrading via the '/opt/tinypilot-updater/venv/bin/python3 -m pip install --upgrade pip' command.
```

I have no idea how to deal with pip/dependencies, so I'd appreciate any advice

mtlynch, 2022-06-27:
Thanks for reporting this!

We'll fix this in #1032. Can you try installing from that branch instead?

```bash
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/no-pkg-resources/quick-install | \
    bash - && \
  sudo reboot
```

spitfire, 2022-06-27:
Thanks for your reply.

Not it's failing compilation, is there an issue for that?

```
Building wheels for collected packages: cryptography
  Building wheel for cryptography (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/tinypilot-updater/venv/bin/python3 /opt/tinypilot-updater/venv/lib/python3.10/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmpnrfbzi3j
       cwd: /tmp/pip-install-wp06w3gl/cryptography_3b5ea3fc9e184829a093d181ea14c7d8
  Complete output (194 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-cpython-310
  creating build/lib.linux-armv7l-cpython-310/cryptography
  copying src/cryptography/utils.py -> build/lib.linux-armv7l-cpython-310/cryptography
  copying src/cryptography/fernet.py -> build/lib.linux-armv7l-cpython-310/cryptography
  copying src/cryptography/exceptions.py -> build/lib.linux-armv7l-cpython-310/cryptography
  copying src/cryptography/__init__.py -> build/lib.linux-armv7l-cpython-310/cryptography
  copying src/cryptography/__about__.py -> build/lib.linux-armv7l-cpython-310/cryptography
  creating build/lib.linux-armv7l-cpython-310/cryptography/x509
  copying src/cryptography/x509/oid.py -> build/lib.linux-armv7l-cpython-310/cryptography/x509
  copying src/cryptography/x509/ocsp.py -> build/lib.linux-armv7l-cpython-310/cryptography/x509
  copying src/cryptography/x509/name.py -> build/lib.linux-armv7l-cpython-310/cryptography/x509
  copying src/cryptography/x509/general_name.py -> build/lib.linux-armv7l-cpython-310/cryptography/x509
  copying src/cryptography/x509/extensions.py -> build/lib.linux-armv7l-cpython-310/cryptography/x509
  copying src/cryptography/x509/certificate_transparency.py -> build/lib.linux-armv7l-cpython-310/cryptography/x509
  copying src/cryptography/x509/base.py -> build/lib.linux-armv7l-cpython-310/cryptography/x509
  copying src/cryptography/x509/__init__.py -> build/lib.linux-armv7l-cpython-310/cryptography/x509
  creating build/lib.linux-armv7l-cpython-310/cryptography/hazmat
  copying src/cryptography/hazmat/_oid.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat
  copying src/cryptography/hazmat/__init__.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat
  creating build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/poly1305.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/padding.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/keywrap.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/hmac.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/hashes.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/constant_time.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/cmac.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/_serialization.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/_cipheralgorithm.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/_asymmetric.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/__init__.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives
  creating build/lib.linux-armv7l-cpython-310/cryptography/hazmat/bindings
  copying src/cryptography/hazmat/bindings/__init__.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/bindings
  creating build/lib.linux-armv7l-cpython-310/cryptography/hazmat/backends
  copying src/cryptography/hazmat/backends/interfaces.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/backends
  copying src/cryptography/hazmat/backends/__init__.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/backends
  creating build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives/twofactor
  copying src/cryptography/hazmat/primitives/twofactor/totp.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives/twofactor
  copying src/cryptography/hazmat/primitives/twofactor/hotp.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives/twofactor
  copying src/cryptography/hazmat/primitives/twofactor/__init__.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives/twofactor
  creating build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives/serialization
  copying src/cryptography/hazmat/primitives/serialization/ssh.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives/serialization
  copying src/cryptography/hazmat/primitives/serialization/pkcs7.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives/serialization
  copying src/cryptography/hazmat/primitives/serialization/pkcs12.py -> build/lib.linux-armv7l-cpython-310/cryptography/hazmat/primitives/serialization
[... cut ...] e the Rust compiler toolchain.

  This package requires Rust >=1.41.0.
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects
WARNING: You are using pip version 21.3.1; however, version 22.1.2 is available.
You should consider upgrading via the '/opt/tinypilot-updater/venv/bin/python3 -m pip install --upgrade pip' command.
```

spitfire, 2022-06-27:
Thanks for the hints.
It's totally understandable for me you can't support something rather exotic as Orange Pi (One). From my point of view it's a perfect device for this, as it has:

- Dedicated power supply (so no worries about powering from computer)
- OTG microUSB port
- USB-A port to pług in the capture card
- Ethernet port to connect to the network

Since it has Armbian that is based on Ubuntu 22.04 (should be fairly close to Debian which RPi OS is based on) I hoped it would be pretty straightforward (maybe aside from hardware acceleration for media which has been spotty on it). 

As far as I can see none of the issues listed in that article (such as an outdated kernel or python) are the case here, any idea what the compilation error is about?

PR #1032 (2022-06-28): Remove pkg-resources from list of quick-install requirements
According to this thread, this line was always erroneous and we should just remove it: https://github.com/pypa/pip/issues/4022

It doesn't seem to harm anything on Raspberry Pi OS, but it causes problems on other distros.

Fixes #1031
Fixes #1015

---
This change is [[image]](https://reviewable.io/reviews/tiny-pilot/tinypilot/1032)