tiny-pilot/tinypilot #1067: Update Overhaul: `e2e` step incorrectly tests previous bundle version

jdeanwallace, 2022-07-25:
The CircleCI `e2e` step basically just [runs `get-tinypilot.sh` and checks if the webserver is returning `HTTP 200`](https://github.com/tiny-pilot/tinypilot/blob/2fd84113cbf3ef57f2f7adf533beff68b7f07b5c/e2e/docker-test#L35-L37). However, the [`e2e` step doesn't wait for the `LATEST` file to be updated](https://app.circleci.com/pipelines/github/tiny-pilot/tinypilot/2384/workflows/7fd79944-6536-4de0-b4b9-c780682948dd) with the next build version before running because that would effectively "release" an untested version which makes the `e2e` step pointless.

jdeanwallace, 2022-07-25:
> I think we just need to add a `requires` clause to `e2e` like this

Are we okay with overwriting the [`LATEST` file](https://github.com/tiny-pilot/tinypilot/blob/2fd84113cbf3ef57f2f7adf533beff68b7f07b5c/.circleci/config.yml#L163-L171) and "releasing" the bundle even if the `e2e` step fails?

mtlynch, 2022-07-25:
@jdeanwallace - Yeah, it's not ideal but adding something like a staging layer is too complicated.

It's the same as what we've been doing pre-update overhaul. By the time the e2e test ran in the previous flow, we had already merged the code into `master` and made it available to users.

closed 2022-07-26 by jdeanwallace