tiny-pilot/tinypilot #1070: Update Overhaul: Unable to upgrade from Community to Pro

jdeanwallace, 2022-07-26:
Related to https://github.com/tiny-pilot/tinypilot/issues/1027

On a device that has already applied the `update-overhaul` changes and is now trying to upgrade to Pro, `ansible-role-tinypilot-pro` is unable to `git pull` the Pro repo because the `/opt/tinypilot` directory is not empty.

Ansible logs:
```bash
TASK [tinypilot.tinypilot-pro : create TinyPilot folder] ***********************
ok: [localhost]

TASK [tinypilot.tinypilot-pro : get TinyPilot repo] ****************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": "/usr/bin/git clone --origin origin REDACTED", "msg": "fatal: destination path '/opt/tinypilot' already exists and is not an empty directory.", "rc": 128, "stderr": "fatal: destination path '/opt/tinypilot' already exists and is not an empty directory.\n", "stderr_lines": ["fatal: destination path '/opt/tinypilot' already exists and is not an empty directory."], "stdout": "", "stdout_lines": []}
```

Directory contents:
```bash
$ ls -la /opt/tinypilot
total 52
drwxr-xr-x  5 tinypilot tinypilot 4096 Jul 26 13:35 .
drwxr-xr-x  7 root      root      4096 Jul 26 13:30 ..
-rw-r--r--  1 tinypilot tinypilot   55 Jul 26 13:23 COPYRIGHT
-rw-r--r--  1 tinypilot tinypilot 1054 Jul 26 13:23 LICENSE
-rw-r--r--  1 tinypilot tinypilot 7902 Jul 26 13:23 README.md
-rw-r--r--  1 tinypilot tinypilot    8 Jul 26 13:26 VERSION
-rw-r--r--  1 tinypilot tinypilot    0 Jul 26 13:23 __init__.py
drwxr-xr-x 10 tinypilot tinypilot 4096 Jul 26 13:36 app
-rw-r--r--  1 tinypilot tinypilot  169 Jul 26 13:09 app_settings.cfg
-rw-r--r--  1 tinypilot tinypilot  404 Jul 26 13:23 requirements.txt
drwxr-xr-x  2 tinypilot tinypilot 4096 Jul 26 13:35 scripts
-rw-r--r--  1 tinypilot tinypilot  468 Jul 26 13:23 setup.py
drwxr-xr-x  6 tinypilot tinypilot 4096 Jul 26 13:35 venv
```

mtlynch, 2022-07-26:
Good catch!

Cleanest solution I can think of is that we add a snippet to [`install.sh` like](https://github.com/tiny-pilot/tinypilotkvm.com/blob/953791861b7f6c2476d8df7507e20297f153d16d/static/pro/install.sh):

```bash
if [[ -d /opt/tinypilot ]]; then
  sudo rm -rf /opt/tinypilot
fi
```

It wipes out `app_settings.cfg`, but we should move that anyway because it shouldn't be in `/opt/tinypilot` and keeps causing issues. For now, it's unlikely that users changed `app_settings.cfg` except to enable H264, but they can re-enable after the upgrade.

jotaen4tinypilot, 2022-07-26:
@jdeanwallace do you happen to know whether this is the only issue, or are there more errors down the line in the installation/upgrade process? (E.g. due to the internal structure of `/opt/tinypilot-updater` being different.)

I’m wondering what the best way would be to go about this, because I think this should only be a temporary problem for the transition period where Community is on the new update flow but Pro is still on the old. (Which should only be 2–3 weeks, optimally.)

This also made me realize that the [“Test upgrade from Community to Pro” testing item](https://github.com/tiny-pilot/tinypilot/issues/1027) should actually be two-fold: a) “upgrade from overhauled Community to legacy Pro”, and b) “upgrade from overhauled Community to overhauled Pro”. We obviously can’t test the latter yet, since Pro is not ready to go yet, so we need to keep track of that.

Crazy question, in case this issue here would turn out to contain more complications: would it be feasible to suspend automatic upgrades from Community to Pro throughout the transition time? (Not sure how exactly that could look then, though…)

jdeanwallace, 2022-07-27:
@jotaen4tinypilot 

> do you happen to know whether this is the only issue, or are there more errors down the line in the installation/upgrade process? (E.g. due to the internal structure of `/opt/tinypilot-updater` being different.)

Not that I know off. After this fix, things seem to run smoothly.

> I’m wondering what the best way would be to go about this, because I think this should only be a temporary problem for the transition period where Community is on the new update flow but Pro is still on the old. (Which should only be 2–3 weeks, optimally.)

Yeah definitely a temporary problem, but the fix will also be temporary because the install script will eventually point to `get-tinypilot-pro.sh` and then install the Debian package which [conditionally wipes `/opt/tinypilot`](https://github.com/tiny-pilot/tinypilot-pro/blob/master/Dockerfile#L52-L56).

> This also made me realize that the [“Test upgrade from Community to Pro” testing item](https://github.com/tiny-pilot/tinypilot/issues/1027) should actually be two-fold: a) “upgrade from overhauled [...]

Good point, I've added a "Final QA testing" issue stub for Pro: https://github.com/tiny-pilot/tinypilot-pro/issues/531

> Crazy question, in case this issue here would turn out to contain more complications: would it be feasible to suspend automatic upgrades from Community to Pro throughout the transition time? (Not sure how exactly that could look then, though…)

I don't think this would be too much of an issue besides cause some user confusion and perhaps a support ticket or two. But I don't think the extra pressure to get Pro out is worth it.

closed 2022-07-29 by mtlynch