tiny-pilot/tinypilot #1073: Development workflow to test changes on device?

jotaen4tinypilot, 2022-07-26:
The [`/dev-scripts/remote-scripts` folder](https://github.com/tiny-pilot/tinypilot/tree/master/dev-scripts/remote-scripts) is currently [referenced in the Contributing guide lines](https://github.com/tiny-pilot/tinypilot/blob/master/CONTRIBUTING.md#remote-scripts).

- First question: (cc @jdeanwallace, @mtlynch) is anyone still using these remote scripts?
  - I personally don’t use them anymore (though I have used them in the beginning).
- Second question: shall we document the default/canonical way to push code changes to the device?
  - I’d say yes.
- Third question: what is the default way to push local code changes to the device?
  - I use `rsync` and [composed a local helper script](https://gist.github.com/jotaen4tinypilot/bbd1f38437c021d93bc8bbe8451e194c) for myself for convenience (I sync from my machine to the device). That worked quite well for me. This will not work with the update overhaul anymore, though, at least not flawlessly, because [we strip a lot of files from the bundle](https://github.com/tiny-pilot/tinypilot/blob/master/.dockerignore). One could argue the excluded files are non-production files anyway, so it wouldn’t matter if they are synced to the device. But it’s still not optimal. A potential solution could be a transient/intermediate docker step, that pushes all files through Docker (and thus `.dockerignore`) before syncing them up.

mtlynch, 2022-07-26:
>First question: (cc @jdeanwallace, @mtlynch) is anyone still using these remote scripts? 

No, I never ended up using these. I don't think I move code to devices enough to have a regular workflow.

>Second question: shall we document the default/canonical way to push code changes to the device? 

Yeah, I think that's a good idea.

>Third question: what is the default way to push local code changes to the device? 

If I were doing it, I think I'd just create a bundle and then install that on the target device, but that's pretty slow.

Your script looks straightforward, so I support making that the official way if Jason doesn't have a way he likes better.

jdeanwallace, 2022-07-27:
> First question

No, I'm not using it anymore.

> Second question

Yeah that would be useful.

> Third question

I haven't really thought about it yet, but your script looks cool. I would also prefer a quick way to bundle up the code and install it the "official" way (i.e. Debian package), but I don't think that's feasible anymore.

Would be cool to a add [watcher type thing](https://eradman.com/entrproject/) to your script and let it sync automagically too 🙌

jotaen4tinypilot, 2022-07-27:
Cool, thanks for your thoughts. I’ve updated the ticket description, and tried to make the ticket actionable.

As always, feel free to edit/adjust/extend the requirements list.

mtlynch, 2022-10-14:
Agreed, I've added it to the next milestone.

mtlynch, 2022-12-15:
I'm going to push this until 2.6.1. The priority right now is getting to Bullseye + audio capture. I'd like to minimize the lag between when our hardware is audio-ready but our software isn't.