tiny-pilot/tinypilot #1077: Allow Reverse Proxying

LukeLambert, 2022-07-28:
Like many, I use a reverse proxy in front of most web services to terminate TLS, provide SSO, and authorize requests (aka Zero Trust). The TinyPilot Nginx config can easily allow for this setup with a few tweaks copied from the [nginx-proxy](https://github.com/nginx-proxy/nginx-proxy) project. It also greatly simplifies the config.

Note: I'm not sure which headers are needed for the /janus/ws location. I haven't observed my TinyPilot using that endpoint.

```Nginx
# If we receive X-Forwarded-Proto, pass it through; otherwise, pass along the
# scheme used to connect to this server
map $http_x_forwarded_proto $proxy_x_forwarded_proto {
  default $http_x_forwarded_proto;
  ''      $scheme;
}

# If we receive X-Forwarded-Port, pass it through; otherwise, pass along the
# server port the client connected to
map $http_x_forwarded_port $proxy_x_forwarded_port {
  default $http_x_forwarded_port;
  ''      $server_port;
}

# If we receive Upgrade, set Connection to "upgrade"; otherwise, delete any
# Connection header that may have been passed to this server
map $http_upgrade $proxy_connection {
  default upgrade;
  '' close;
}

server {
    listen       80 default_server;
    listen       [::]:80 default_server;

    server_name tinypilot;

    root /opt/tinypilot;

    index index.html;
    
    proxy_buffers 16 16k;
    proxy_buffer_size 16k;
    proxy_set_header Host $http_host;
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $proxy_connection;
    proxy_set_header X-Real-IP $remote_addr;
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header X-Forwarded-Proto $proxy_x_forwarded_proto;
    proxy_set_header X-Forwarded-Port $proxy_x_forwarded_port;
    proxy_http_version 1.1;

    location /state {
      proxy_pass http://ustreamer;
    }
    location /stream {
      postpone_output 0;
      proxy_buffering off;
      proxy_ignore_headers X-Accel-Buffering;
      proxy_pass http://ustreamer;
    }
    location /snapshot {
      proxy_pass http://ustreamer;
    }
    location /janus/ws {
      proxy_pass http://janus-ws;
      proxy_set_header X-Scheme $proxy_x_forwarded_proto;
    }
    location / {
      proxy_pass http://tinypilot;
    }
    location ~* ^/.+\.(html|js|js.map|css|woff|woff2)$ {
      root "/opt/tinypilot/app/static";

      # We cache assets to prevent the browser from making redundant
      # requests to the same files while loading the page. (Observed on
      # Chrome 91.) We don’t want caching otherwise, though, in order to
      # avoid stale files after users update their device. Note, that in
      # addition to `max-age`, the browser’s caching behaviour is relative
      # to the `Last-Modified` header, so we make that seem recent.
      add_header Last-Modified $date_gmt;
      add_header Cache-Control 'public, max-age=10s';
    }
    location ~* ^/.+\.(jpg|jpeg|png|ico)$ {
      root "/opt/tinypilot/app/static";
    }
}
```

mtlynch, 2022-07-28:
Thanks for reaching out!

Can you clarify the request a bit? Is this a feature request or are you sharing the config that you've created?

LukeLambert, 2022-07-28:
Ultimately I'm requesting a feature: the ability to use a reverse proxy in front of TinyPilot. This is just an example config that should allow that without impacting any existing functionality. (Though I'm not sure what `janus` is or how it might be impacted by different headers.) I'm not familiar with the TinyPilot codebase, but I didn't find the current Nginx config in this repository.

mtlynch, 2022-07-28:
The nginx config is here:

https://github.com/tiny-pilot/ansible-role-tinypilot/blob/14cc40e7e53260628be09597799533e080bfc7f9/tasks/nginx.yml

Janus is a WebRTC server. It's necessary if you're using [H264 video](https://tinypilotkvm.com/blog/whats-new-in-2022-05#h264-video-over-webrtc), but if you're using the default (MJPEG), you don't need to worry about it.

I'm still not quite clear on the feature request. TinyPilot does work with a reverse proxy already. Is the request to document how to set up a reverse proxy in front of TinyPilot?

This thread might be helpful: https://forum.tinypilotkvm.com/-174

LukeLambert, 2022-07-28:
I'll read through that thread. I was never able to get an upstream reverse proxy working with the current config. The problem is that Nginx ignores `X-Forwarded-Proto` and `X-Forwarded-Port` from any upstream reverse proxies. Since Nginx is receiving http on port 80, it passes those as the protocol and port to the Flask app. When accessing TinyPilot from an origin such as `https://tinypilot.example.com` it will generate a CORS error.

LukeLambert, 2022-07-28:
Yes, I believe that user was experiencing the same issue because Flask thought it was being accessed over http on port 80. Werkzeug ProxyFix inspects the X-Forwarded-For headers to construct redirects and check CORS.

This config would solve that. It's generic and should work with any upstream reverse proxy. I have several apps that go Traefik -> Nginx -> Flask. Sometimes Cloudflare is even in front of Traefik. As long as each proxy passes the correct X-Forwarded-For and Host headers, everything works as expected.

mtlynch, 2022-07-28:
Ah, okay. 

Can you make this a PR into the ansible-role-tinypilot repo?

https://github.com/tiny-pilot/ansible-role-tinypilot/blob/14cc40e7e53260628be09597799533e080bfc7f9/tasks/nginx.yml

closed 2022-07-28 by mtlynch

LukeLambert, 2022-07-28:
I tried the config with a few reverse proxies and they all worked.

### ngrok
```
docker run -it --rm -e NGROK_AUTHTOKEN=AUTHTOKEN ngrok/ngrok http TINYPILOT_IP
```
You'll need a free [ngrok authtoken](https://dashboard.ngrok.com/get-started/your-authtoken). Open the HTTPS address printed in the terminal.
**Warning: This is publicly accessible.**

### Cloudflare Tunnel

In the [teams dashboard](https://dash.teams.cloudflare.com), under Access > Tunnels, click *Create a tunnel* and follow the prompts. Open the HTTPS address of the subdomain you created.
**Warning: This is publicly accessible.**

### Caddy
```
docker run -it --rm -p 8080:80 caddy caddy reverse-proxy -from :80 -to TINYPILOT_IP
```
Open `http://localhost:8080`.

### Caddy HTTPS
```
docker run -it --rm -p 8443:443 caddy caddy reverse-proxy -to TINYPILOT_IP
```
Open `https://localhost:8443` and ignore the security warnings.