tiny-pilot/tinypilot #1078: Update Overhaul: Parity of `quick-install` and the new bundler

jotaen4tinypilot, 2022-07-30:
As part of the [final update overhaul QA](https://github.com/tiny-pilot/tinypilot/issues/1027), I compared all instructions from `quick-install` with our new bundling mechanism, and discovered the following differences. As far as I know, none of these things has caused any issues so far, though.

- We don’t check for the legacy [`$TINYPILOT_INSTALL_VARS`](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/quick-install#L31-L35) anymore. That check should probably go into `get-tinypilot.sh`.
  - On the other hand, we could also decide that it’s about time to drop this check now?
- Previously, [we created `ansible.cfg`](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/quick-install#L164-L167) on the fly. The `$PWD` variable in line 165 was automatically resolved by bash. Now, [the file is “hard coded”](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/bundler/bundle/ansible.cfg) in the bundler folder. Consequently, `$PWD` isn’t resolved anymore, but it appears literally in the file.
  - Apparently, this hasn’t created any issues so far, so we might be able to drop this line altogether?
- We don’t run [`ansible-galaxy install`](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/quick-install#L179-L181) anymore on the `ansible-role-tinypilot` folder.

Other than the above, we have ported everything.

jdeanwallace, 2022-08-01:
Thanks @jotaen4tinypilot 

> We don’t check for the legacy [$TINYPILOT_INSTALL_VARS](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/quick-install#L31-L35) anymore.

I'm in favor of dropping the check. We've deprecated the use of `$TINYPILOT_INSTALL_VARS` [more than a year ago](https://github.com/tiny-pilot/tinypilot/commit/650ac357219a0d84cef4e422ad61d847a0fdc2d0).

> Consequently, `$PWD` isn’t resolved anymore, but it appears literally in the file.

I also stumbled upon this before and thought it was a bug. I can't find any docs on using $PWD inside the `ansible.cfg` file, except for a few [random comments about it on github](https://github.com/ansible/ansible/issues/5185#issuecomment-458095127). I tested it for myself, and it does seem to expand `$PWD` correctly.

> We don’t run [ansible-galaxy install](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/quick-install#L179-L181) anymore on the ansible-role-tinypilot folder.

Oh this is interesting. Currently we [manually `git pull` the dependent ansible roles](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/bundler/create-bundle#L57-L61). If we wanted to we could probably replace that entire part with:
```bash
ansible-galaxy install git+https://github.com/tiny-pilot/ansible-role-tinypilot.git,update-overhaul
```
which should download all nested dependencies too (i.e. `ansible-role-nginx`, `ansible-role-nginx`). This might be a better solution to avoid worrying about maintaining nested dependencies.

mtlynch, 2022-08-02:
>>We don’t check for the legacy [$TINYPILOT_INSTALL_VARS](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/quick-install#L31-L35) anymore.

>I'm in favor of dropping the check. We've deprecated the use of $TINYPILOT_INSTALL_VARS [more than a year ago](https://github.com/tiny-pilot/tinypilot/commit/650ac357219a0d84cef4e422ad61d847a0fdc2d0).

Yeah, we can drop that at this point.

>>Consequently, $PWD isn’t resolved anymore, but it appears literally in the file.

>I also stumbled upon this before and thought it was a bug. I can't find any docs on using $PWD inside the ansible.cfg file, except for a few https://github.com/ansible/ansible/issues/5185#issuecomment-458095127. I tested it for myself, and it does seem to expand $PWD correctly.

I'd rather not rely on undocumented behavior, but I feel like it's easy to fix as part of the other issue below, so I've filed https://github.com/tiny-pilot/tinypilot/issues/1081 to track.

>>We don’t run [ansible-galaxy install](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/quick-install#L179-L181) anymore on the ansible-role-tinypilot folder.

>Oh this is interesting. Currently we [manually git pull the dependent ansible roles](https://github.com/tiny-pilot/tinypilot/blob/1a72ab42d27141fce9aa241126b6bad8029ee1f4/bundler/create-bundle#L57-L61). If we wanted to we could probably replace that entire part with:
>
>`ansible-galaxy install git+https://github.com/tiny-pilot/ansible-role-tinypilot.git,update-overhaul`
>
>which should download all nested dependencies too (i.e. ansible-role-nginx, ansible-role-nginx). This might be a better solution to avoid worrying about maintaining nested dependencies.

I filed https://github.com/tiny-pilot/tinypilot/issues/1082 to track this.

I wouldn't think this is worth fixing right now, but I realized we forgot to create a work item for [preserving explicit role versions](https://github.com/tiny-pilot/tinypilot-pro/issues/535) in TinyPilot Pro, and #1082 will make that easier, so we might as well fix it now.

closed 2022-08-02 by mtlynch