tiny-pilot/tinypilot #1099: Revise HDMI to CSI instructions so they work on top of an existing install

mtlynch, 2022-09-12:
We recently were [helping a customer in the support forum](https://forum.tinypilotkvm.com/-383#post-9) who had an HDMI to CSI setup, but they had run the installation for HDMI to USB.

We should adjust our [HDMI to CSI install instructions](https://github.com/tiny-pilot/tinypilot/wiki/Installation-Options#example-tc358743-hdmi-to-csi-capture-chip) so that they work in both scenarios:

* A bare Raspbian system
* A Raspbian system where the user installed TinyPilot Community with the default options

db39, 2022-10-06:
This is the bash snippet that fixed the user's issue after they had already installed using the simple setup:

```bash
echo 'ustreamer_capture_device: tc358743' | sudo tee /home/tinypilot/settings.yml
```
It overwrites `settings.yml` with the capture device definition. The user then has to run the `get-tinypilot.sh` script again to complete the setup.

To make the HDMI → CSI instructions work in both scenarios, I think we'll need to separate the creation of `settings.yml` (with the device definition) and the `tinypilot` user. Otherwise, the current snippet will fail for users who have already installed with `get-tinypilot.sh`.

I think it might be a good idea to try and preserve the other settings too. The default install leaves `settings.yml` like this:
```bash
ustreamer_port: 8001
ustreamer_persistent: true
ustreamer_encoder: hw
ustreamer_format: jpeg
ustreamer_resolution: 1920x1080
```
`ustreamer_encoder: hw` needs to be changed to `ustreamer_encoder omx` or removed so that it gets set up properly in `get-tinypilot.sh` (it's only set if it's undefined):

```bash
yaml_set_if_undefined /var/tmp/tmp.o0wJuR5v00.yml ustreamer_encoder omx
```

Overwriting the whole file (leaving just the capture device definition) works. But that will affect previously defined custom settings (e.g., `ustreamer_resolution: "1280x720"`).

So would it be better to do something like this:

```bash
echo 'ustreamer_capture_device: tc358743' | sudo tee -a /home/tinypilot/settings.yml && \
    sed --in-place --expression "/ustreamer_encoder/d" /home/tinypilot/settings.yml
```

which preserves those settings?

mtlynch, 2022-10-06:
By default, [we set three settings](https://github.com/tiny-pilot/tinypilot/blob/b7c726299cb0062f18cbc4ed555b183c2b168408/bundler/bundle/install#L86L88) if it's not TC358743:

```bash
yaml_set_if_undefined "${INSTALL_SETTINGS_FILE}" "ustreamer_encoder" "hw"
yaml_set_if_undefined "${INSTALL_SETTINGS_FILE}" "ustreamer_format" "jpeg"
yaml_set_if_undefined "${INSTALL_SETTINGS_FILE}" "ustreamer_resolution" "1920x1080"
```

So the bash snippet should add `ustreamer_capture_device: tc358743` and delete the three defaults.

db39, 2022-10-06:
Thanks for the clarification - I overlooked that. 

What if we did something like this? 

```bash
TINYPILOT_SETTINGS="/home/tinypilot/settings.yml" && \
if [ -f "${TINYPILOT_SETTINGS}" ]; then
  sudo sed --in-place \
  --expression "/ustreamer_capture_device/d" \
  --expression "/ustreamer_encoder/d" \
  --expression "/ustreamer_format/d" \
  --expression "/ustreamer_resolution/d" ${TINYPILOT_SETTINGS} && \
  echo 'ustreamer_capture_device: tc358743' | sudo tee -a "${TINYPILOT_SETTINGS}"
else 
  sudo useradd --create-home --system --user-group tinypilot && \
  echo 'ustreamer_capture_device: tc358743' | sudo tee -a "${TINYPILOT_SETTINGS}" && \
  sudo chown tinypilot:tinypilot "${TINYPILOT_SETTINGS}"
fi
```

It checks whether `settings.yml` exists, then edits or creates it with the correct content. So it works if user runs it on a bare Raspian system, or if they already ran `get-tinypilot.sh`.

mtlynch, 2022-10-06:
That looks good. A couple small notes:

We follow Google's [bash conventions](https://google.github.io/styleguide/shellguide.html#test----and---), so it should be double brackets (`if [[ -f ...`])

We indent for line continuations, so it should be:

```bash
TINYPILOT_SETTINGS="/home/tinypilot/settings.yml" && \
if [[ -f "${TINYPILOT_SETTINGS}" ]]; then
  sudo sed --in-place \
    --expression "/ustreamer_capture_device/d" \
    --expression "/ustreamer_encoder/d" \
    --expression "/ustreamer_format/d" \
    --expression "/ustreamer_resolution/d" ${TINYPILOT_SETTINGS} && \
    echo 'ustreamer_capture_device: tc358743' | sudo tee -a "${TINYPILOT_SETTINGS}"
else 
  sudo useradd --create-home --system --user-group tinypilot && \
    echo 'ustreamer_capture_device: tc358743' | sudo tee -a "${TINYPILOT_SETTINGS}" && \
    sudo chown tinypilot:tinypilot "${TINYPILOT_SETTINGS}"
fi
```

db39, 2022-10-07:
Thanks for the reminder on those points. Are you happy for me to edit the wiki page with that new command?

mtlynch, 2022-10-07:
Yep, it's good to go on the wiki.

db39, 2022-10-07:
Done!

mtlynch, 2022-10-07:
Excellent, thanks!

closed 2022-10-07 by mtlynch