tiny-pilot/tinypilot #1106: Include menu items for features that require manual action

mtlynch, 2022-09-20:
We have several network operations that I eventually want to integrate in the web GUI, but they're currently things you have to perform manually with SSH.

* [Set a static IP](https://tinypilotkvm.com/faq/static-ip)
* [Enable WiFI](https://tinypilotkvm.com/faq/enable-wifi)
* Specify an HTTP proxy (coming soon)

It would be good to help the user discover that those features are available even if we don't yet have a smooth web UI for managing them.

The challenge here is to balance competing forces:

* We want to make TinyPilot's functionality discoverable
* We don't want to clutter the UI by mixing features that work as expected and features that feel half-baked and tell the user to go do it themselves on the command-line

Is there a way to achieve this? Some potential ideas:

* Add a menu item like "Manual features" that has dropdowns to the things that require the user to do it manually
* Add a tag like "alpha" or "experimental" to the menu item to convey that it's still in progress

mtlynch, 2022-11-11:
Sorry for the delay on this! It's been a busy couple of weeks, but this is still on my radar to review.

jdeanwallace, 2022-12-13:
@mtlynch - Just checking to see if this is still on your radar to review?

mtlynch, 2023-05-10:
Sorry, this keeps reaching the top of my to-do list and I think, "Oh, I'm going to have to think hard about this, but it's not urgent," and I keep pushing it.

>Instructions
>
>How exactly do we want to make the instructions available?

I think linking out to the FAQ is the best option. I'd like to have it be in a place where we can continuously update it. The `?version` query parameter is a good idea, and we should do that since it's easy to do and gives us flexibility in the future.

>Another thing would be whether we need to provide the beta instructions forever, since it might happen that a user is still on an ancient version, even though the then-beta feature had long been released for real in the meantime. On the one hand, we could be so kind to preserve old instructions; on the other hand, we could take that opportunity and tell users “this is a real feature now, please just update your device”.

I'm not *too* worried about this. It's pretty easy for us to just update the text to say "Update your device" and then archive the FAQ page so it doesn't show up in the FAQ index.

>Therefore, I think that we should use this opportunity to revisit the menu structure (of the “System” menu mainly), in order to properly account for the growing number of items and their different character. It would make sense to me to work on that in parallel, because I see cross-cutting concerns with whatever we do here.

I agree, but we also want to minimize churn. If we, for example, moved "Logs" to a different heading, then we suddenly have dozens of support forum posts that become incorrect for future versions.

>we could consider to actively draw the user’s attention to a new beta feature after they performed a system update. So if the new TinyPilot version comes with a newly available beta-feature, we could show an indicator to make the user aware of the additional functionality. Since not all users might be interested in beta-features, such a notification mechanism could be opt-in.

I wouldn't call these beta features exactly because that implies that we're planning to polish them, and we're just sharing an early draft. But for some of these features, we just added manual instructions because a few users asked, but they're not worth the implementation cost / maintenance burden of building a web UI for them.

I feel like if we present them as "here are some exciting new beta features!" the user will expect that we're on the road to making them first-class features. Maybe it's a distinction between "highlighting" the feature and just making it discoverable. Because I think our goal is to make it discoverable rather than generating excitement. We just want to prevent a user from discovering through some other means that a static IP is technically possible, and then they think, "Why didn't TinyPilot tell me I could do that?"

shalver-tp, 2024-05-17:
Re-opening this in general, but also for implementation of some potential new menu items.

shalver-tp, 2024-05-21:
Some of the thinking on this issue is potentially outdated, so I will just try to recap and reset a bit.

For the most part, we do not want to change the location of anything if we can avoid it. We also do not want to add more menus, nor do we want to add more items to an individual menu - _but_ there are some items that we want to integrate as a key part of the product and as such they need a menu item.

### Current Menus

The 2.6.3 Menu Tree

- **System** - settings for the TinyPilot device's system
   - **Security** - opens popup for Web Access, SSH Access, and require HTTPS
   - **Virtual Media** - opens popup for managing virtual media
   - **Update** - opens popup and checks for updates
   - **Networking**
      - **Hostname** - allows you to change the hostname of device
      - **Static IP** - allows you to set a static IP for device
   - **Video Settings** - opens popup to change video stream settings
   - **Logs** - opens popup with log information
   - **Power** - opens popup with options to shut down or restart
- **Actions** - this is actually very well described, these are actions that the user can launch
   - **Paste** - opens popup and allows you to paste text to send to your target computer
   - **Screenshot** - takes and saves a screenshot of the target computer
   - **Wake on LAN** - opens a popup to allow you to send a WOL signal to another device on the network
   - **Keyboard Shortcuts**
      - **Ctrl Alt Del** - allows you to send Ctrl Alt Del to the target computer
- **View** - options for how you view TinyPilot
   - **Cursor** - options for customizing the appearance of your cursor
      - Default
      - None
      - Crosshair
      - Pointer
   - **Show Keyboard** - Check on or off to show or hide on-screen keyboard
   - **Enable Key History** - Check on or off to enable key history
   - **Full Screen** - when selected, your target computer goes Full Screen
   - **Dedicated Window** - when selected your target computer pops out into a different window
- **Help**
   - **About** - popup that provides version, license, and privacy information
   - **Technical Support** - pops out a browser tab for tinypilotkvm.com/contact/
   - **FAQ** - pop outs a browser tab for tinypilotkvm.com/faq

### New Menu Items

There are a number of items that currently require a code snippet (or similar) to enable but have enough frequency in customer support interactions or are a new "feature" that we should try to surface them from a menu.

- Configure Wi-Fi
- Launch Console
- Mouse Jiggler
- Enable/Disable Read-Only Filesystem

#### Configure Wi-Fi - This can go under **System** -> **Networking** -> (below Static IP)
[FAQ](https://tinypilotkvm.com/faq/enable-wifi/), Related Issue: [131](https://github.com/tiny-pilot/tinypilot/issues/131)

#### Launch Console - This can go under **Actions** (below Wake on LAN)
Related Issue: [1332](https://github.com/tiny-pilot/tinypilot-pro/issues/1332)

#### Mouse Jiggler - This can go under **Actions** (below Launch Console)
[FAQ](https://tinypilotkvm.com/faq/mouse-jiggler)

#### Enable/Disable Read-Only Filesystem - This can go in **System** -> (below Update)
[FAQ](https://tinypilotkvm.com/faq/read-only-filesystem), Related Issue: [1331](https://github.com/tiny-pilot/tinypilot-pro/issues/1331)

shalver-tp, 2024-06-03:
@jotaen4tinypilot I think this ticket is potentially obsolete, the only thing worth preserving or linking to might be the discussion beginning from this comment. https://github.com/tiny-pilot/tinypilot/issues/1106#issuecomment-2121528254