tiny-pilot/tinypilot #1110: Remove explicit uStreamer parameters from bundle installer

mtlynch, 2022-09-28:
The bundle install script manually specifies 

https://github.com/tiny-pilot/tinypilot/blob/59e73d15f456cfcb31703937c0a48d2c297da335/bundler/bundle/install#L78L82

These settings shouldn't be necessary, as uStreamer will pick default settings if we don't specify anything as long as we've set `capture_device: "tc358743"`:

https://github.com/tiny-pilot/ansible-role-ustreamer/blob/490b8b46c06b1fb613c07e2d6e8bf5a315cf619c/tasks/provision_tc358743.yml#L61L74

This doesn't really cause a problem now, but when we switch to Raspbian Bullseye and uStreamer 5.x, the OMX encoder is no longer available, so it's better to let uStreamer dynamically choose between OMX and M2M depending on the uStreamer version and the OS version.

clarkewd, 2022-12-26:
I think I can confirm that this is broken. On a fresh image of Bullseye I first run this:

    TINYPILOT_SETTINGS="/home/tinypilot/settings.yml" &&   sudo useradd --create-home --system --user-group tinypilot &&   echo 'ustreamer_capture_device: tc358743' | sudo tee -a "${TINYPILOT_SETTINGS}" &&   sudo chown tinypilot:tinypilot "${TINYPILOT_SETTINGS}"

Followed by the install curl:

    curl \
      --silent \
      --show-error \
      https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh | \
        bash - && \
      sudo reboot

I get this error during setup:

    TASK [ansible-role-ustreamer : check that OpenMax IL acceleration is only compiled on Raspberry Pi OS 10 (Buster) and earlier] **************************
    fatal: [localhost]: FAILED! => {"changed": false, "msg": "OpenMax IL acceleration (OMX) is only available on Raspberry Pi OS 10 (Buster) and earlier."}

Is there a fix or workaround?

mtlynch, 2022-12-27:
Sorry about that! 

As a workaround, can you try running this bash snippet before the `curl` command?

```bash
sudo sed --in-place '/ustreamer_encoder: /d' \
    /home/tinypilot/settings.yml && \
  echo 'ustreamer_encoder: m2m-image' \
    | sudo tee --append /home/tinypilot/settings.yml
```

mtlynch, 2023-01-17:
@jotaen4tinypilot - Yeah, that's a good idea. I don't recall clearly how we ended up with all the settings defined in two places, but it's likely one or more of:

* I wasn't thinking things through
* It was time-consuming to test which would still work if we removed them
* I was worried about breaking something legacy if I removed them

mtlynch, 2023-01-18:
Oh, neat!