tiny-pilot/tinypilot #1120: Use underscore prefix consistently for private JS members

jotaen4tinypilot, 2022-09-30:
According to our coding conventions, all (supposed-to-be) [private JS members should have an underscore prefix](https://github.com/tiny-pilot/tinypilot/blob/6073998459e21317800139df38166cd34396d4fd/CONTRIBUTING.md?plain=1#L214-L216).

- I think we are fairly consistent in regards to naming methods here, but it would be good to double-check this anyway
- For fields, we have non-prefixed ones, [e.g. in `<remote-screen>`](https://github.com/tiny-pilot/tinypilot/blob/6073998459e21317800139df38166cd34396d4fd/app/templates/custom-elements/remote-screen.html#L114). So we should check all components.
- The conventions around our common identifiers [`this.state`](https://github.com/tiny-pilot/tinypilot/blob/6073998459e21317800139df38166cd34396d4fd/CONTRIBUTING.md?plain=1#L144-L146) and [`this.elements`](https://github.com/tiny-pilot/tinypilot/blob/6073998459e21317800139df38166cd34396d4fd/CONTRIBUTING.md?plain=1#L216) need to be updated to `this._state` and `this._elements`, and all code refactored.

We also have to do this in Pro.

jdeanwallace, 2024-03-27:
From a recent meeting with @mtlynch, to keep this task a `small` we've limited the scope of this task to the following PRs:
* https://github.com/tiny-pilot/tinypilot/pull/1758
* https://github.com/tiny-pilot/tinypilot/pull/1759
* https://github.com/tiny-pilot/tinypilot/pull/1760
* https://github.com/tiny-pilot/tinypilot/pull/1761

The outstanding changes are less uniform and might be tedious/time-consuming to review.

jdeanwallace, 2024-03-28:
Completed via
* https://github.com/tiny-pilot/tinypilot/pull/1758
* https://github.com/tiny-pilot/tinypilot/pull/1759
* https://github.com/tiny-pilot/tinypilot/pull/1760
* https://github.com/tiny-pilot/tinypilot/pull/1761

closed 2024-03-28 by jdeanwallace