tiny-pilot/tinypilot #1140: Ubuntu 22.04: ustreamer version not showing up in logs

billdenney, 2022-10-11:
## Description

Related to #1134

I was looking at the log file, and the ustreamer version did not show up.

## What's the behavior that you expect?

ustreamer version is in the log file

## What's happening instead?

No ustreamer verison in the log

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Click on "System > Logs"
3. See ustreamer version without logs

## Screenshots

See the third line below

```
Software versions
TinyPilot version: b7c7262
uStreamer version:  
OS version: Linux tinypilot 5.15.0-1016-raspi #18-Ubuntu SMP PREEMPT Wed Sep 28 12:15:55 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
Distribution name: Ubuntu
Distribution version: 22.04
```

## Logs

https://logs.tinypilotkvm.com/DKMtIYga

mtlynch, 2022-10-11:
Did you reinstall uStreamer?

The versioning is just [checking the git tags](https://github.com/tiny-pilot/ansible-role-tinypilot/blob/1dabf92ce53ab46384206e6bf7ab493542e8da9e/files/collect-debug-logs#L74) for the `/opt/ustreamer` directory, so I'd expect them to be there unless you wiped the directory and installed uStreamer in a different way.

billdenney, 2022-10-11:
I didn't do anything specific for uStreamer.  I just did the default installation from the script.  Now that I know where to look, I think that the issue relates to a directory detection issue:

```
bill@tinypilot:/opt/ustreamer$ git branch
fatal: detected dubious ownership in repository at '/opt/ustreamer'
To add an exception for this directory, call:

        git config --global --add safe.directory /opt/ustreamer
```

That appears to be related to https://github.blog/2022-04-12-git-security-vulnerability-announced/

So, while it doesn't affect your typical production environment, the commands will with an upgrade to git.  This work-around should help:

`git config --global --add safe.directory '/opt/ustreamer'`

That would need to be run as the tinypilot user.

billdenney, 2022-10-11:
Correction:  The above git config would need to be run for root.  Then it works:

```
Software versions
TinyPilot version: b7c7262
uStreamer version: v5.23 90f09b1
OS version: Linux tinypilot 5.15.0-1016-raspi #18-Ubuntu SMP PREEMPT Wed Sep 28 12:15:55 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
Distribution name: Ubuntu
Distribution version: 22.04
```

mtlynch, 2022-10-17:
Oh, yeah, that security fix [bit us in other ways](https://github.com/tiny-pilot/tinypilot/issues/963).

We should just get the version number from the binary itself. Fixing in: https://github.com/tiny-pilot/ansible-role-tinypilot/pull/231

billdenney, 2022-10-17:
That makes sense.  Feel free to close this when you're ready to.

closed 2022-10-19 by mtlynch