tiny-pilot/tinypilot #1169: Ability to disable keyboard and/or mouse interaction

cole-h, 2022-10-28:
It would be swell if I could disable TinyPilot from forwarding my keypresses to the KVM, and require using the software keyboard on the TinyPilot page to send keyboard events. Same with the mouse, it would be nice to disable forwarding mouse movement to the KVM with some kind of toggle.

cole-h, 2022-10-31:
Yes, sorry for the lack of details!

The issue I'm facing is this:

1. Visit the TinyPilot web interface
2. Watch the connected device do its things
3. Press Ctrl+Tab to go to the next tab in my browser
4. Notice that TinyPilot also received (and forwarded) those keys before switching away

The case where this was specifically annoying was when trying to test something once the attached machine went to sleep, but of course switching back and forth to that tab with keyboard shortcuts just kept it awake for longer.

In the end, it's not a terribly annoying issue (I don't expect to have to do it very frequently, so I can just remember to not use keyboard shortcuts there :P), but I wonder if it would be possible and/or worthwhile to require the virtualized display be focused in order to capture the keyboard and mouse (or have a setting to do so, or something similar).

mtlynch, 2022-10-31:
One other feature we're considering that might solve this would be to have a read-only URL (e.g. http://tinypilot/read-only?key=abcd1234), where you can see the remote screen but you can't send keyboard/mouse input or change settings.

Would that work for your scenario? When you wanted to just view the screen, you could visit the read-only URL, but when you wanted to send keyboard/mouse input, you could visit the normal interface.

cole-h, 2022-11-01:
I think that would serve me well indeed! I hope it would also come with a link between the two so I don't have to remember if it's `/read-only` or `/readonly` or similar 😆