tiny-pilot/tinypilot #1182: Installation Fails at "build ustreamer"

saltyfishychips, 2022-11-06:
## Description

Briefly summarize the problem you're experiencing.

## What's the behavior that you expect?

To be installed without issue.

## What's happening instead?

TASK [ansible-role-ustreamer : build uStreamer] ************************************************************************fatal: [localhost]: FAILED! => {"changed": false, "cmd": "/usr/bin/gmake WITH_OMX=0 WITH_JANUS=1", "msg": "In file included from src/audio.c:23:\nsrc/audio.h:34:10: fatal error: alsa/asoundlib.h: No such file or directory\n   34 | #include <alsa/asoundlib.h>\n      |          ^~~~~~~~~~~~~~~~~~\ncompilation terminated.\ngmake[2]: *** [Makefile:40: build/src/audio.o] Error 1\ngmake[1]: *** [Makefile:49: janus] Error 2\ngmake: *** [Makefile:29: all] Error 2", "rc": 2, "stderr": "In file included from src/audio.c:23:\nsrc/audio.h:34:10: fatal error: alsa/asoundlib.h: No such file or directory\n   34 | #include <alsa/asoundlib.h>\n      |          ^~~~~~~~~~~~~~~~~~\ncompilation terminated.\ngmake[2]: *** [Makefile:40: build/src/audio.o] Error 1\ngmake[1]: *** [Makefile:49: janus] Error 2\ngmake: *** [Makefile:29: all] Error 2\n", "stderr_lines": ["In file included from src/audio.c:23:", "src/audio.h:34:10: fatal error: alsa/asoundlib.h: No such file or directory", "   34 | #include <alsa/asoundlib.h>", "      |          ^~~~~~~~~~~~~~~~~~", "compilation terminated.", "gmake[2]: *** [Makefile:40: build/src/audio.o] Error 1", "gmake[1]: *** [Makefile:49: janus] Error 2", "gmake: *** [Makefile:29: all] Error 2"], "stdout": "/usr/bin/gmake apps\ngmake[1]: Entering directory '/opt/ustreamer'\n/usr/bin/gmake -C src\ngmake[2]: Entering directory '/opt/ustreamer/src'\n-- CC libs/base64.c\n-- CC libs/frame.c\n-- CC libs/logging.c\n-- CC libs/memsink.c\n-- CC libs/options.c\n-- CC libs/unjpeg.c\n-- CC ustreamer/blank.c\n-- CC ustreamer/data/blank_jpeg.c\n-- CC ustreamer/data/index_html.c\n-- CC ustreamer/device.c\n-- CC ustreamer/encoder.c\n-- CC ustreamer/encoders/cpu/encoder.c\n-- CC ustreamer/encoders/hw/encoder.c\n-- CC ustreamer/h264/stream.c\n-- CC ustreamer/http/bev.c\n-- CC ustreamer/http/mime.c\n-- CC ustreamer/http/path.c\n-- CC ustreamer/http/server.c\n-- CC ustreamer/http/static.c\n-- CC ustreamer/http/unix.c\n-- CC ustreamer/http/uri.c\n-- CC ustreamer/m2m.c\n-- CC ustreamer/main.c\n-- CC ustreamer/options.c\n-- CC ustreamer/stream.c\n-- CC ustreamer/workers.c\n== LD ustreamer.bin\n-- CC dump/file.c\n-- CC dump/main.c\n== LD ustreamer-dump.bin\ngmake[2]: Leaving directory '/opt/ustreamer/src'\ngmake[1]: Leaving directory '/opt/ustreamer'\n/usr/bin/gmake janus\ngmake[1]: Entering directory '/opt/ustreamer'\n/usr/bin/gmake -C janus\ngmake[2]: Entering directory '/opt/ustreamer/janus'\n-- CC src/audio.c\ngmake[2]: Leaving directory '/opt/ustreamer/janus'\ngmake[1]: Leaving directory '/opt/ustreamer'\n", "stdout_lines": ["/usr/bin/gmake apps", "gmake[1]: Entering directory '/opt/ustreamer'", "/usr/bin/gmake -C src", "gmake[2]: Entering directory '/opt/ustreamer/src'", "-- CC libs/base64.c", "-- CC libs/frame.c", "-- CC libs/logging.c", "-- CC libs/memsink.c", "-- CC libs/options.c", "-- CC libs/unjpeg.c", "-- CC ustreamer/blank.c", "-- CC ustreamer/data/blank_jpeg.c", "-- CC ustreamer/data/index_html.c", "-- CC ustreamer/device.c", "-- CC ustreamer/encoder.c", "-- CC ustreamer/encoders/cpu/encoder.c", "-- CC ustreamer/encoders/hw/encoder.c", "-- CC ustreamer/h264/stream.c", "-- CC ustreamer/http/bev.c", "-- CC ustreamer/http/mime.c", "-- CC ustreamer/http/path.c", "-- CC ustreamer/http/server.c", "-- CC ustreamer/http/static.c", "-- CC ustreamer/http/unix.c", "-- CC ustreamer/http/uri.c", "-- CC ustreamer/m2m.c", "-- CC ustreamer/main.c", "-- CC ustreamer/options.c", "-- CC ustreamer/stream.c", "-- CC ustreamer/workers.c", "== LD ustreamer.bin", "-- CC dump/file.c", "-- CC dump/main.c", "== LD ustreamer-dump.bin", "gmake[2]: Leaving directory '/opt/ustreamer/src'", "gmake[1]: Leaving directory '/opt/ustreamer'", "/usr/bin/gmake janus", "gmake[1]: Entering directory '/opt/ustreamer'", "/usr/bin/gmake -C janus", "gmake[2]: Entering directory '/opt/ustreamer/janus'", "-- CC src/audio.c", "gmake[2]: Leaving directory '/opt/ustreamer/janus'", "gmake[1]: Leaving directory '/opt/ustreamer'"]}

PLAY RECAP *************************************************************************************************************localhost                  : ok=22   changed=1    unreachable=0    failed=1    skipped=23   rescued=0    ignored=0

## What are the steps to reproduce this behavior?

1. Connect pi via SSH
2. Run simple installation command

## Logs

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`
N/A, could not install

julianborghuis, 2022-11-06:
I have sadly the same error. How to solve this?

saltyfishychips, 2022-11-07:
Fixed by manually installing the missing files. The following worked for me:

``
sudo apt-get install libasound2-dev

sudo apt-get install build-essential libspeex-dev libspeexdsp-dev libpulse-dev

sudo apt-get install libopus-dev

mtlynch, 2022-11-07:
Thanks for reporting this! The workaround is to install these packages:

```bash
sudo apt install -y libasound2-dev libspeex-dev libspeexdsp-dev libopus-dev
```

But the installer should be doing this for you, so we need to figure out what's going wrong.

mtlynch, 2022-11-07:
Okay, this was fixed in https://github.com/tiny-pilot/ansible-role-ustreamer/pull/71 but we hadn't generated a new install bundle since Thursday. I just manually forced a new install bundle publish, so this should be fixed now.

closed 2022-11-07 by mtlynch