tiny-pilot/tinypilot #1189: Mouse don't work on BIOS

eleksbai, 2022-11-10:
## Description

TinyPilot's keyboard and mouse don't work properly on HP laptop's bios, only the keyboard works

## What's the behavior that you expect?

Mouse works fine

## What's happening instead?

Keyboard works fine, mouse doesn't work

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Press esc on HP laptop to enter the StartUP Menu
3. At this time the web keyboard works but the web mouse does not work
4. Click F10 on the webpage to enter BIOS SETUP
5. At this time the web keyboard works but the web mouse does not work

## Screenshots

[image]
[image]
[image]

## Logs

https://logs.tinypilotkvm.com/AzORi1XW

mtlynch, 2022-11-10:
Thanks for reporting this!

Which model HP laptop is this?

If you plug a physical USB mouse into the laptop, does it work? Are there any BIOS options for enabling external mouse input? Some laptops are picky about what external devices they'll interact with in BIOS.

eleksbai, 2022-11-10:
Thank Bro!
my laptop is HP ProBook 440 G7 ,And my physical USB mouse into the laptop can work.

eleksbai, 2022-11-10:
Thank Bro!
My laptop is HP ProBook 440 G7 ,And my physical USB mouse into the laptop can work.

> Thanks for reporting this!
> 
> Which model HP laptop is this?
> 
> If you plug a physical USB mouse into the laptop, does it work? Are there any BIOS options for enabling external mouse input? Some laptops are picky about what external devices they'll interact with in BIOS.

closed 2022-11-10 by eleksbai