tiny-pilot/tinypilot #1205: Log whether the read-only filesystem is enabled in debug logs

mtlynch, 2022-11-21:
In some cases, users get confused that settings revert after reboot because they've forgotten that they've enabled the overlay filesystem / read-only filesystem.

We should include in the debug logs an indicator of whether the overlay filesystem is enabled.

mtlynch, 2022-11-30:
@cghague - Can you see if you can find a way to determine whether or not the overlay filesystem is enabled or not from within a bash script?

cghague, 2022-11-30:
@mtlynch - No problem! This should do the trick:

```bash
#!/bin/bash

grep -q "boot=overlay" /proc/cmdline
if [ $? -eq 0 ] ; then
  echo "The root partition is read-only!"
fi

findmnt /boot | grep -q " ro,"
if [ $? -eq 0 ] ; then
  echo "The boot partition is read-only!"
fi
```

The irony is that after testing this script on a read-only system I briefly wondered where it had gone after I rebooted! :rofl:

mtlynch, 2022-11-30:
@cghague - Awesome! Could you submit a PR to integrate this into [the log collection script](https://github.com/tiny-pilot/tinypilot/blob/3c4193bb41748d58ce43cd6535bd52b11d1fccd9/debian-pkg/opt/tinypilot-privileged/scripts/collect-debug-logs)?

I'd like to make it so that the log file looks like this:

#### Enabled

```text
...
TinyPilot state
Read-only filesystem: on
temp=49.1'C
...
```

### Disabled

```text
...
TinyPilot state
Read-only filesystem: off
temp=49.1'C
...
```

PR #1218 (2022-12-06): Added step to check if overlay filesystem is enabled
Resolves #1205 by adding a step to the log collection script to check if the overlay / read-only filesystem is enabled.

[image]
commit 2b8cc1e0 (2022-12-06): Added step to check if overlay filesystem is enabled (#1218)

Resolves #1205 by adding a step to the log collection script to check if
the overlay / read-only filesystem is enabled.

<a data-ca-tag
href="https://codeapprove.com/pr/tiny-pilot/tinypilot/1218"><img
src="https://codeapprove.com/external/github-tag-allbg.png" alt="Review
on CodeApprove" /></a>
closed 2022-12-06 by cghague