tiny-pilot/tinypilot #1211: H264 video streaming setting do not work on RPI 4

funkymotions, 2022-11-29:
## Description

Switching to H264 video streaming mode seems not work

## What's the behavior that you expect?

Choosing H264 streaming mode works correct.

## What's happening instead?

Switching btw MJPEG to H264 do not work properly. Seems that H264 streaming setting do not apply at all because UI reports a fallback to MJPEG streaming settings.

## What are the steps to reproduce this behavior?

1. Open TinyPilot web UI
2. Click on "Video Setting"
3. Select H264 streaming setting
4. Apply settings
5. H264 does not properly work as UI reports (see screenshot below)

## Screenshots
[image]

## Logs
https://logs.tinypilotkvm.com/duGoM55e

mtlynch, 2022-11-29:
Thanks for reporting this, @funkymotions!

I think the issue is here:

```text
ExecStart=/opt/ustreamer/ustreamer \
  --host 127.0.0.1 \
  --port 8001 \
  --encoder hw \   <<< Passing through MJPEG video
  --format jpeg \  <<< without encoding for H264
  --resolution 1920x1080 \
  --persistent \
  --h264-sink tinypilot::ustreamer::h264 \
  --h264-sink-mode 777 \
  --h264-sink-rm \
  --h264-bitrate 10000 \
```

I suspect that for H264 + format=jpeg is not a valid combination since uStreamer needs to re-encode the video for H264, so we probably need to change the encoder to CPU and maybe the format as well.

jdeanwallace, 2022-12-08:
I'm struggling to recreate this issue. H264 seems to be working fine on my Raspberry Pi 4 running TinyPilot Community:

[image]

I made sure to match the software version and video settings of the OP. Here are my logs: https://logs.tinypilotkvm.com/O8lRmcOTh

Looking at the uStreamer logs:
```
Dec 08 15:02:51 raspberrypi ustreamer[1621]: -- ERROR [1379.537    stream] -- Could not obtain the requested pixelformat=JPEG; driver gave us MJPEG
Dec 08 15:02:51 raspberrypi ustreamer[1621]: -- INFO  [1379.538    stream] -- Falling back to pixelformat=MJPEG
Dec 08 15:02:51 raspberrypi ustreamer[1621]: -- INFO  [1379.538    stream] -- Using pixelformat: MJPEG
```
In the above error, uStreamer seems to failover to MJPEG. We can avoid this non-critical error by setting `--format mjpeg`.

> I suspect that for H264 + format=jpeg is not a valid combination since uStreamer needs to re-encode the video for H264, so we probably need to change the encoder to CPU and maybe the format as well.

Changing the `encoder` to CPU doesn't do anything and HW is still used because the `format` takes precedence over the `encoder`:
[image]

mtlynch, 2022-12-08:
Oh, that's interesting. Maybe it was a temporary hardware failure?

@funkymotions - Are you still seeing this behavior?

mtlynch, 2022-12-15:
Closing as nonreproducible due to lack of response.

closed 2022-12-15 by mtlynch