tiny-pilot/tinypilot #1212: install script fails at: ansible-role-ustreamer : install Janus Debian package fails

magore, 2022-11-29:
Description:
On a clean raspberry pi install with recommened prerequsits. The install script task fails at the step ansible-role-ustreamer : install Janus Debian package

Install script used:
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/get-tinypilot.sh
 | \  
   bash - && \
sudo reboot

Log file excerpt at point of first failure: 
(note the missing deb package: janus_1.0.1-20221104_armhf27_7a3m1.deb)
...
TASK [ansible-role-ustreamer : check that the H264 variables are in a consistent state] ***
skipping: [localhost]

TASK [ansible-role-ustreamer : determine whether to install Janus] *************
ok: [localhost]

TASK [ansible-role-ustreamer : install Janus Debian package] *******************

TASK [ansible-role-ustreamer : install Janus Debian package] *******************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "dpkg --force-confdef --force-confold -i /root/.ansible/tmp/ansible-tmp-1669750816.8810685-1996-25743789793470/janus_1.0.1-20221104_armhf27_7a3m1.deb failed", "stderr": "dpkg: error processing archive /root/.ansible/tmp/ansible-tmp-1669750816.8810685-1996-25743789793470/janus_1.0.1-20221104_armhf27_7a3m1.deb (--install):\n trying to overwrite '/usr/lib/arm-linux-gnueabihf/libnice.so.10', which is also in package libnice10:armhf 0.1.16-1\nErrors were encountered while processing:\n /root/.ansible/tmp/ansible-tmp-1669750816.8810685-1996-25743789793470/janus_1.0.1-20221104_armhf27_7a3m1.deb\n", "stderr_lines": ["dpkg: error processing archive /root/.ansible/tmp/ansible-tmp-1669750816.8810685-1996-25743789793470/janus_1.0.1-20221104_armhf27_7a3m1.deb (--install):", " trying to overwrite '/usr/lib/arm-linux-gnueabihf/libnice.so.10', which is also in package libnice10:armhf 0.1.16-1", "Errors were encountered while processing:", " /root/.ansible/tmp/ansible-tmp-1669750816.8810685-1996-25743789793470/janus_1.0.1-20221104_armhf27_7a3m1.deb"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following package was automatically installed and is no longer required:\n  libfuse2\nUse 'sudo apt autoremove' to remove it.\nThe following NEW packages will be installed:\n  libconfig9\n0 upgraded, 1 newly installed, 0 to remove and 96 not upgraded.\nNeed to get 27.7 kB of archives.\nAfter this operation, 66.6 kB of additional disk space will be used.\nGet:1 http://raspbian.freemirror.org/raspbian bullseye/main armhf libconfig9 armhf 1.5-0.4 [27.7 kB]\nFetched 27.7 kB in 0s (60.3 kB/s)\nSelecting previously unselected package libconfig9:armhf.\r\n(Reading database ... \r(Reading database ... 5%\r(Reading database ... 10%\r(Reading database ... 15%\r(Reading database ... 20%\r(Reading database ... 25%\r(Reading database ... 30%\r(Reading database ... 35%\r(Reading database ... 40%\r(Reading database ... 45%\r(Reading database ... 50%\r(Reading database ... 55%\r(Reading database ... 60%\r(Reading database ... 65%\r(Reading database ... 70%\r(Reading database ... 75%\r(Reading database ... 80%\r(Reading database ... 85%\r(Reading database ... 90%\r(Reading database ... 95%\r(Reading database ... 100%\r(Reading database ... 102406 files and directories currently installed.)\r\nPreparing to unpack .../libconfig9_1.5-0.4_armhf.deb ...\r\nUnpacking libconfig9:armhf (1.5-0.4) ...\r\nSetting up libconfig9:armhf (1.5-0.4) ...\r\nProcessing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u4) ...\r\nSelecting previously unselected package janus.\n(Reading database ... 102414 files and directories currently installed.)\nPreparing to unpack .../janus_1.0.1-20221104_armhf27_7a3m1.deb ...\nUnpacking janus (1.0.1) ...\nProcessing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u4) ...\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following package was automatically installed and is no longer required:", "  libfuse2", "Use 'sudo apt autoremove' to remove it.", "The following NEW packages will be installed:", "  libconfig9", "0 upgraded, 1 newly installed, 0 to remove and 96 not upgraded.", "Need to get 27.7 kB of archives.", "After this operation, 66.6 kB of additional disk space will be used.", "Get:1 http://raspbian.freemirror.org/raspbian bullseye/main armhf libconfig9 armhf 1.5-0.4 [27.7 kB]", "Fetched 27.7 kB in 0s (60.3 kB/s)", "Selecting previously unselected package libconfig9:armhf.", "(Reading database ... ", "(Reading database ... 5%", "(Reading database ... 10%", "(Reading database ... 15%", "(Reading database ... 20%", "(Reading database ... 25%", "(Reading database ... 30%", "(Reading database ... 35%", "(Reading database ... 40%", "(Reading database ... 45%", "(Reading database ... 50%", "(Reading database ... 55%", "(Reading database ... 60%", "(Reading database ... 65%", "(Reading database ... 70%", "(Reading database ... 75%", "(Reading database ... 80%", "(Reading database ... 85%", "(Reading database ... 90%", "(Reading database ... 95%", "(Reading database ... 100%", "(Reading database ... 102406 files and directories currently installed.)", "Preparing to unpack .../libconfig9_1.5-0.4_armhf.deb ...", "Unpacking libconfig9:armhf (1.5-0.4) ...", "Setting up libconfig9:armhf (1.5-0.4) ...", "Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u4) ...", "Selecting previously unselected package janus.", "(Reading database ... 102414 files and directories currently installed.)", "Preparing to unpack .../janus_1.0.1-20221104_armhf27_7a3m1.deb ...", "Unpacking janus (1.0.1) ...", "Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u4) ..."]}
to help explain your problem._

magore, 2022-11-29:
Full log from install script
[tinypilotlog.txt](https://github.com/tiny-pilot/tinypilot/files/10117023/tinypilotlog.txt)

mtlynch, 2022-11-29:
Thanks for reporting this @magore! This is a known issue:

https://github.com/tiny-pilot/ansible-role-ustreamer/issues/75

The workaround is to run `apt remove -y libnice10` and then re-run the installer.

We're working on a fix so this doesn't fail in the first place.

closed 2022-11-29 by mtlynch