tiny-pilot/tinypilot #1236: Errors during install on 64-bit Raspberry Pi OS

billdenney, 2022-12-19:
Related to #929

## Description

I was trying to install on 64-bit Raspberry Pi OS (commenting out the stop due to 64-bit in the installer).  During installation, I got the following two errors:

```
TASK [ansible-role-ustreamer : check if legacy Janus package is installed] **********************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": "dpkg-query --show --showformat='${Maintainer}' janus | grep TinyPilot", "delta": "0:00:00.025175", "end": "2022-12-19 16:17:43.595620", "msg": "non-zero return code", "rc": 1, "start": "2022-12-19 16:17:43.570445", "stderr": "dpkg-query: no packages found matching janus", "stderr_lines": ["dpkg-query: no packages found matching janus"], "stdout": "", "stdout_lines": []}
...ignoring
```

```
TASK [ansible-role-ustreamer : add archive apt repository key] **********************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to download key at https://ftp-master.debian.org/keys/archive-key-11.asc: Request failed: <urlopen error [Errno 113] No route to host>"}
```

The second error appears to be related to a temporary error with the Debian mirror server where it appears to have been down for about 2 days (starting on 2022-12-17T17:45).

## What's the behavior that you expect?

No errors during installation.

## What's happening instead?

The two above errors occurred.

## What are the steps to reproduce this behavior?

1. run `cd /opt/tinypilot-updater; sudo install'

## Screenshots

Not applicable

billdenney, 2022-12-19:
For the second issue, I'm not sure that there's much you can do to fix a Debian server that is down.  There may be a different way to get to the keys as described here:  https://wiki.debian.org/SecureApt#How_to_find_and_add_a_key

I don't know if it's reasonable to do that as a fallback or if it's more complexity than it's worth.

mtlynch, 2022-12-19:
The first issue is expected and the installer correctly ignores it. The second one, yeah, I'm not sure if there's anything to do except wait for Debian to get their server back online. I'm trying to find any information about the outage.

mtlynch, 2022-12-19:
Found the outage thread: https://lists.debian.org/debian-infrastructure-announce/2022/12/msg00002.html

Courtesy of Vincent Bernat: https://m.mtlynch.io/@vbernat@hachyderm.io/109541365767265861

billdenney, 2022-12-19:
... and the server is back up now (https://mirror-master.debian.org/status/mirror-info/ftp-master.debian.org.html).  It may be worth considering using the `debian-archive-keyring` package instead, but I got farther in my installation.

I'll close the issue when the installation step is complete (which should be in just a few minutes, I'd guess).

billdenney, 2022-12-19:
Installation completed after the server was back up.

closed 2022-12-19 by billdenney