tiny-pilot/tinypilot #1237: Bad H.264 Encoding

billdenney, 2022-12-19:
## Description

While using my recently installed system, I had an encoding issue with H.264.  The bottom lines of the screen flickered green and white.

My Video settings are H.264, 30 fps, 2 Mb/s.  Resetting the encoder to 5 Mb/s did not fix the issue.  Also, changing to MJPEG did not fix the issue.

## What's the behavior that you expect?

Good encoding.

## What's happening instead?

Bad encoding

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Type ls
3. See bad encoding at the bottom of the screen

The system itself was still usable.  I could type and have changes occur, but they were mostly unreadable.

## Screenshots

[image]

## Logs

[Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`](https://logs.tinypilotkvm.com/Keru1CcC)

mtlynch, 2022-12-19:
Thanks for reporting this!

We have very limited dev resources, so we only support 32-bit Raspbian. Users are welcome to install and contribute bugfixes on whatever system they want, but we don't have resources to investigate issues outside of 32-bit Raspbian.

Can you try reproducing on 32-bit?

billdenney, 2022-12-21:
I will give it a shot to either give 32-bit a try or otherwise find a fix.

With a first look at the logs, it appears to be an issue with ustreamer or upstream of that (i.e. maybe the driver or firmware for the HDMI capture device):

```
Dec 19 18:31:02 tinypilot ustreamer[2575]: Corrupt JPEG data: 7 extraneous bytes before marker 0xd9
Dec 19 18:31:02 tinypilot ustreamer[2575]: Corrupt JPEG data: premature end of data segment
```

billdenney, 2022-12-21:
After a bit of investigation, it could be related to pikvm/ustreamer#169 (which may be subsequently related to a kernel bug).  Is there a simple way to update tinypilot to use ustreamer to 5.34 to test that fix?

mtlynch, 2022-12-21:
Thanks for digging into this! I've made a PR to upgrade to the latest.

If you want to test it early, you can run this snippet on your TinyPilot:

```bash
cd $(mktemp -d)
wget https://output.circle-artifacts.com/output/job/44515202-eb79-4753-a035-1e330fcd9302/artifacts/0/bundler/dist/tinypilot-community-20221221T1645Z-1.8.0-63+07f25a0.tgz
tar -xvf tinypilot-community-20221221T1645Z-1.8.0-63+07f25a0.tgz
./install
```

billdenney, 2022-12-21:
Thanks for the update!

I tested the new version of ustreamer, and it didn't fix the issue.  (I confirmed in the logs that it was updated.)

mtlynch, 2022-12-22:
Ah, darn! Is it possibly defective hardware? Are you using an HDMI to USB dongle? Do you have another on hand that you can try?

billdenney, 2022-12-22:
I am using an HDMI to USB dongle, and I just tried another one (from the same manufacturer, that's the only type I have), and I'm getting the same result.

mtlynch, 2022-12-22:
Can you share a link to the HDMI dongle? Is it [one of these](https://github.com/tiny-pilot/tinypilot/wiki/HDMI-Capture-Devices#macrosilicon-ms2109-capture-devices)?

Can you share the logs from when you just run it with MJPEG and skip the H264 stuff? 

```bash
sudo service ustreamer stop # Temporarily turn off the service

/opt/ustreamer/ustreamer \
  --host :: \
  --port 8001 \
  --encoder hw \
  --format jpeg \
  --resolution 1920x1080 \
  --persistent
```

billdenney, 2022-12-23:
Yes, it's the second of the HDMI dongles on your list: https://smile.amazon.com/gp/product/B08CXWPYQ8

The ustreamer logs are pretty quiet:

```
$ /opt/ustreamer/ustreamer \
  --host :: \
  --port 8001 \
  --encoder hw \
  --format jpeg \
  --resolution 1920x1080 \
  --persistent
-- INFO  [19280.218      main] -- Starting PiKVM uStreamer 5.34 ...
-- INFO  [19280.218      main] -- Using internal blank placeholder
-- INFO  [19280.219      main] -- Listening HTTP on [::]:8001
-- INFO  [19280.219    stream] -- Using V4L2 device: /dev/video0
-- INFO  [19280.219    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [19280.225      http] -- Starting HTTP eventloop ...
-- INFO  [19280.291    stream] -- Device fd=8 opened
-- INFO  [19280.291    stream] -- Using input channel: 0
-- INFO  [19280.292    stream] -- Using resolution: 1920x1080
-- ERROR [19280.292    stream] -- Could not obtain the requested format=JPEG; dr                                                                                 iver gave us MJPEG
-- INFO  [19280.292    stream] -- Falling back to format=MJPEG
-- INFO  [19280.293    stream] -- Using format: MJPEG
-- INFO  [19280.294    stream] -- Using HW FPS: 0 -> 60 (coerced)
-- ERROR [19280.294    stream] -- Device doesn't support setting of HW encoding                                                                                  quality parameters
-- INFO  [19280.294    stream] -- Using IO method: MMAP
-- INFO  [19280.319    stream] -- Requested 5 device buffers, got 5
-- INFO  [19280.328    stream] -- Capturing started
-- INFO  [19280.328    stream] -- Using JPEG quality: encoder default
-- INFO  [19280.328    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [19280.328    stream] -- Capturing ...
-- INFO  [19306.798      http] -- HTTP: NEW client (now=1): [192.168.0.156]:39510, id=e4b31ca92b6d1c77
```

It seems not to like something about this filename when printed to the screen on a black background with green foreground:

`fwupd-94d6edee-8f19-40a9-be14-d881e2e3d55c.cap`

It is fine with that text in black and white on the terminal (by echoing it).

This command will cause that to print with a green foreground (and I have a default of a black background).  When I type the command, I reproducibly get the glitch: 

```
echo -e "\e[1;32m fwupd-94d6edee-8f19-40a9-be14-d881e2e3d55c.cap \e[0m"
```

mtlynch, 2022-12-23:
Hmm, yeah, it sounds like it's either a kernel bug or a uStreamer bug.

You could file a bug with uStreamer, but I'm not sure if they'll have bandwidth to investigate. I'd recommend showing a minimal command-line to trigger the issue, which I guess would be:

```bash
sudo /opt/ustreamer/ustreamer \
  --host :: \
  --port 8001 \
  --encoder hw \
  --format jpeg \
  --resolution 1920x1080 \
  --persistent \
  --h264-sink ustreamer::h264 \
  --h264-sink-mode 777 \
  --h264-sink-rm
```

It might also be useful to test on 32-bit to see if this is limited to 64-bit mode.

billdenney, 2022-12-23:
I just submitted it to ustreamer.  I also tried to grab the raw output from `v4l2-ctl`, but I couldn't figure out the correct command to use for that.

billdenney, 2022-12-27:
I have confirmed that this is an issue with the firmware on the MS2109 device that I'm using (for more detail, see the ustreamer issue linked above).

To confirm that it is a firmware issue, I have switched to using the uncompressed YUYV stream from the card, and there is no issue.  If someone else finds this issue or a similar encoding issue, they can try something like the following:

```
sudo /opt/ustreamer/ustreamer \
  --host :: \
  --port 8001 \
  --encoder cpu \
  --format yuyv \
  --resolution 1920x1080 \
  --persistent
```

I sent an email to MacroSilicon to ask if there is a firmware update, but I'm not holding my breath.

closed 2022-12-27 by billdenney

mtlynch, 2022-12-27:
Ah, bummer. Thanks for investigating this!