tiny-pilot/tinypilot #1244: Add scripts for setting and unsetting a static IP address

mtlynch, 2022-12-30:
In our [static IP address FAQ](https://tinypilotkvm.com/faq/static-ip), we include a convenience script for setting the static IP. The script is inaccessible to users who have TinyPilots without Internet access, so we should ship the script with TinyPilot software itself.

The scripts should be in [`debian-pkg/opt/tinypilot-privileged/scripts/`](https://github.com/tiny-pilot/tinypilot/tree/566c6dc825cfb85996c6c608016e1ff9ae43e62c/debian-pkg/opt/tinypilot-privileged/scripts). They should have these semantics:

```bash
set-static-ip --router router_ip --ip static_ip [--dns dns_ip]

Sets a static local IP address for the TinyPilot device.

--router router_ip: The IP address of the local router.
--ip static_ip: The static IP address to assign to the TinyPilot device.
--dns dns_ip: (optional) The DNS server for TinyPilot to use to resolve domain names. To specify multiple servers, separate the IPs with spaces. If unspecified, defaults to the router's DNS, then Google (8.8.8.8), then Cloudflare (1.1.1.1).

Examples:

# Set static IP to 10.0.105 on a 10.0.0.x network.
set-static-ip \
  --router 10.0.0.1 \
  --ip 10.0.0.105

# Set static IP to 192.168.1.100 and specifies a Quad9 DNS server
set-static-ip \
  --router 192.168.1.1 \
  --ip 192.168.1.100 \
  --dns 9.9.9.9
```

```bash
unset-static-ip

Reverts changes made by the set-static-ip script.
```

The scripts can either be bash scripts or Python scripts.

cghague, 2023-01-04:
@mtlynch - We have instructions on the website for enabling Wi-Fi as an alternative to Ethernet but the current convenience script only works for `eth0`. Should I add an optional `--interface <wlan0>` argument to the script? This will also have the added benefit of future proofing the script to work with other adapters and bridges.

mtlynch, 2023-01-04:
Yeah, that's a good idea!

mtlynch, 2023-01-09:
>Directly call the unset-static-ip script from the set-static-ip script

This sounds like the cleanest.

There are a few places where we have to determine the location of the script, so we have a standard [one-liner](https://github.com/tiny-pilot/tinypilot/blob/49a888f404d0393c635d94f443eec33260913a4a/dev-scripts/enable-mock-scripts#L16) we can reuse.

PR #1256 (2023-01-12): Add scripts to manage static IP addresses
Resolves #1244 by adding scripts to manage static IP addresses.
[image]
commit 98f29a14 (2023-01-12): Add scripts to manage static IP addresses (#1256)

Resolves #1244 by adding scripts to manage static IP addresses.
<a data-ca-tag
href="https://codeapprove.com/pr/tiny-pilot/tinypilot/1256"><img
src="https://codeapprove.com/external/github-tag-allbg.png" alt="Review
on CodeApprove" /></a>
closed 2023-01-12 by cghague