tiny-pilot/tinypilot #1249: Take advantage of uStreamer launcher

mtlynch, 2023-01-04:
Once we merge in https://github.com/tiny-pilot/ansible-role-ustreamer/pull/89 TinyPilot can change uStreamer runtime settings without relying on re-running the uStreamer Ansible role.

We should get rid of `debian-pkg/opt/tinypilot-privileged/scripts/update-video-settings` and just restart the uStreamer and Janus services, as the launcher will read the settings file at runtime.

Incomplete proof-of-concept: https://github.com/tiny-pilot/tinypilot-pro/pull/701

mtlynch, 2023-01-04:
@jdeanwallace - Can you pause the audio work at the next convenient stopping point and take this on? It would be cool if we could get this in with the 2.5.1 release because it makes the video settings experience much faster, though we can punt if it looks like it's going to delay the release by more than a day or two.

mtlynch, 2023-01-06:
@jdeanwallace - What are your thoughts in terms of timing? It'd be fun to slip this into 2.5.1, but it's now the last work item, so I don't want to drag things out too long. How much work do you think is left to implement and test this?

jdeanwallace, 2023-01-09:
@mtlynch - I'm pretty sure I can complete the implementation (and personal testing) today, but then it still would have to be reviewed (by either you or Jan). So best case it could probably be merged by tomorrow.

jdeanwallace, 2023-01-09:
@mtlynch - Sorry on second thought, I think this would delay the `2.5.1` release by too much. The JS code to determine if uStreamer/Janus is back online (after a restart) is trickier than expected.

mtlynch, 2023-01-09:
Okay, good to know. I'll cut a new release candidate and start the test process with that.

closed 2023-01-13 by mtlynch